thonny/thonny #1779: Not working in High Sierra

aivarannamaa, 2021-04-23:
> not working with my high sierra. before it worked well but suddenly it has stopped to working .. no functioning the button (commands of thonny)
> even alvarannamaa's copy also the same

aivarannamaa, 2021-04-23:
@gurugr, I created a new issue for your problem.

Do I understand correctly, that you can start Thonny and see its window, but the buttons are not working?

Can you send me a screenshot of your Thonny window?

gurugr, 2021-04-23:
[
[image]
](url)

gurugr, 2021-04-23:
it is like stuck. not running. when I move it , it's become completely black. i remove it many times restart the computer and reinstall the Tonney but it is the same. i can not understand why it is like this. before it worked  perfectly 
thank you so much for spending time for me.

aivarannamaa, 2021-04-23:
Please post me your Thonny's configuration file:

Open finder, go to your home directory => Library => Thonny. There should be a file named configuration.ini. Please post it under this GitHub issue (either by copy-pasting or by attaching).

gurugr, 2021-04-26:
Thank You so much for your effect. I don't know how, but thonny works now suddenly today. but I will attach to you that configuration.ini file here with a screenshot of my directory

 [@@](`[url]([url](url))`)
[general]
configuration_creation_timestamp = 2021-03-01T14:33:49.505761
language = en_US

[view]
full_screen = False
maximize_view = False
shellview.visible = True
assistantview.visible = True
astview.visible = False
stackview.visible = False
exceptionview.visible = False
filesview.visible = False
heapview.visible = False
helpview.visible = False
notesview.visible = False
objectinspector.visible = True
outlineview.visible = True
variablesview.visible = True
show_program_arguments = False
show_plotter = False

[file]
recent_files = ['/Users/imac/Desktop/wew.py', '/Users/imac/Desktop/python/Angela Lessons/first.py']
current_file = /Users/imac/Desktop/wew.py
open_files = ['/Users/imac/Desktop/wew.py']

[run]
working_directory = /Users/imac/Desktop
program_arguments = 
dock_user_windows = False
pgzero_mode = False

[layout]
zoomed = False
notebook_nw_visible_view = None
notebook_w_visible_view = None
notebook_sw_visible_view = None
notebook_s_visible_view = ShellView
notebook_ne_visible_view = VariablesView
notebook_e_visible_view = None
notebook_se_visible_view = ObjectInspector
width = 833
height = 663
left = 611
top = 435
west_pw_width = 255
east_pw_width = 381
nw_nb_height = 255
sw_nb_height = 255
s_nb_height = 285
se_nb_height = 255
ne_nb_height = 255

[image]

aivarannamaa, 2021-04-26:
I've never tried using emojis in my code. I know that earlier versions of Thonny's UI framework (Tk) did have problems with this.

Can you now use emojis without problems?

gurugr, 2021-04-26:
I've copied a program to see an error with emoji. but the problem was I couldn't erase the code from Thonny. when I check it today thonny has reset (without any previous code) Now I check with a simple code. it worked as usual thank you so much. I guess I've got that error with emojis.

closed 2021-06-13 by aivarannamaa