thonny/thonny #1780: boot.py in raspberrypi pico

amin76hosseini, 2021-04-25:
hi , i write program in boot .py an programed in raspberry pico and i cant program again , how ican delete boot.py ?

CableGuy67, 2021-04-25:
I see what you mean and that seems like it is a bit painful. I set my boot.py to include a file that runs a graphic demo and I had to mash the ctrl-c and hit stop once the demo stopped to finally get the REPL prompt back again. Can you interrupt your running code that way so you can get back to the REPL prompt?
When I tried to just repeatedly stop it I just kept going back to the boot and restarting the demo code from boot.py. I had thought I ran into this before and broke out of it by hitting the stop button a few times but maybe I used ctrl-c that time as well.

aivarannamaa, 2021-04-27:
@amin76hosseini, why you can't program again? Did you get an error message? Can't help you without more details.

katurov, 2021-04-27:
Probably you did "bad code" I've done also. Here is two options (please try 'em). First of all: press "boot" button during the connect - you should see your pico is ok. If it is ok, the problem is in file inself. So here are options: to save code or not to save. See my instruction: https://github.com/katurov/unbrickpipicopython

aivarannamaa, 2021-06-13:
@amin76hosseini, please reopen if you have more details to add!

closed 2021-06-13 by aivarannamaa