thonny/thonny #1781: No module named thonny.plugins.cpython.main

gregsonar, 2021-04-27:
I have a `main.py` script with some basic Flask application

```python
from flask import Flask, render_template, request
import logging
from logging.handlers import RotatingFileHandler
from secrets import token_urlsafe
from flask_socketio import SocketIO, emit
from threading import Lock
import requests
import json  # The imports are redundant, but I plan to use them further 

app = Flask(__name__)
app.config['SECRET_KEY'] = token_urlsafe()

if __name__ == "__main__":
    app.run(debug=True)
```

When I try to run it in Thonny I get error `<...>\AppData\Local\Programs\Thonny\python.exe: No module named thonny.plugins.cpython.main`. When I run it in PyCharm CE, there is no error.

Expected behavior (running Flask app with built-in webserver):
```python
 * Serving Flask app "main" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: on
 * Restarting with stat
2021-04-27 14:37:47,416 - werkzeug - WARNING -  * Debugger is active!
 * Debugger is active!
 * Debugger PIN: 229-074-907
 * Running on http://127.0.0.1:5000/ (Press CTRL+C to quit)

```

What could be the reason for this problem?

aivarannamaa, 2021-04-27:
Thank you for the report!

I suspect you have several Thonny versions on your computer. Please select "Tools => Manage plug-ins" and locate `thonny` package. You should see its installation location. Is it under `<...>\AppData\Local\Programs\Thonny`? if not then you can uninstall this instance.

gregsonar, 2021-04-28:
Thanks for your answer!
The package is installed, displayed in the menu and there are files in this directory.
The IDE itself is also launched from `<...>\AppData\Local\Programs\Thonny`

As far as I can know, this is the only installation on this computer

aivarannamaa, 2021-04-28:
Sorry, I was mislead by a similar looking earlier bug report.

It looks like Flask's reloader is confused by Thonny's approach of starting the process. I'm not sure what's the good solution, but as work around you can either add `use_reloader=False` to the `run`-call or run your script with "Run => Run current script in terminal", which will start the process in a more traditional way.

closed 2022-03-24 by aivarannamaa