thonny/thonny #1782: NameError: name '__thonny_helper' isn't defined

California0306, 2021-04-27:
Hello, I am trying to run code below from pico and I got an error like in title, weeks ago this didn't happend. Some other programs run normally.
What I am missing or doing wrong?
RPi side: 
`import smbus
import time
bus = smbus.SMBus(1)
bus.write_byte(0x41,12)
time.sleep(1)
bus.write_byte(0x41,15)
`
Pico side:
`# pico toggle onboard LED on-off

import machine
from machine import mem32,mem8,Pin
led_onboard = machine.Pin(25, machine.Pin.OUT)

class i2c_slave:
    I2C0_BASE = 0x40044000
    I2C1_BASE = 0x40048000
    IO_BANK0_BASE = 0x40014000
    
    mem_rw =  0x0000
    mem_xor = 0x1000
    mem_set = 0x2000
    mem_clr = 0x3000
    
    IC_CON = 0
    IC_TAR = 4
    IC_SAR = 8
    IC_DATA_CMD = 0x10
    IC_RAW_INTR_STAT = 0x34
    IC_RX_TL = 0x38
    IC_TX_TL = 0x3C
    IC_CLR_INTR = 0x40
    IC_CLR_RD_REQ = 0x50
    IC_CLR_TX_ABRT = 0x54
    IC_ENABLE = 0x6c
    IC_STATUS = 0x70
    
    def write_reg(self, reg, data, method=0):
        mem32[ self.i2c_base | method | reg] = data
        
    def set_reg(self, reg, data):
        self.write_reg(reg, data, method=self.mem_set)
        
    def clr_reg(self, reg, data):
        self.write_reg(reg, data, method=self.mem_clr)
                
    def __init__(self, i2cID = 0, sda=0,  scl=1, slaveAddress=0x41):
        self.scl = scl
        self.sda = sda
        self.slaveAddress = slaveAddress
        self.i2c_ID = i2cID
        if self.i2c_ID == 0:
            self.i2c_base = self.I2C0_BASE
        else:
            self.i2c_base = self.I2C1_BASE
        
        # 1 Disable DW_apb_i2c
        self.clr_reg(self.IC_ENABLE, 1)
        # 2 set slave address
        # clr bit 0 to 9
        # set slave address
        self.clr_reg(self.IC_SAR, 0x1ff)
        self.set_reg(self.IC_SAR, self.slaveAddress &0x1ff)
        # 3 write IC_CON  7 bit, enable in slave-only
        self.clr_reg(self.IC_CON, 0b01001001)
        # set SDA PIN
        mem32[ self.IO_BANK0_BASE | self.mem_clr |  ( 4 + 8 * self.sda) ] = 0x1f
        mem32[ self.IO_BANK0_BASE | self.mem_set |  ( 4 + 8 * self.sda) ] = 3
        # set SLA PIN
        mem32[ self.IO_BANK0_BASE | self.mem_clr |  ( 4 + 8 * self.scl) ] = 0x1f
        mem32[ self.IO_BANK0_BASE | self.mem_set |  ( 4 + 8 * self.scl) ] = 3
        # 4 enable i2c 
        self.set_reg(self.IC_ENABLE, 1)

    def anyRead(self):
        status = mem32[ self.i2c_base | self.IC_RAW_INTR_STAT] & 0x20
        if status :
            return True
        return False

    def put(self, data):
        # reset flag       
        self.clr_reg(self.IC_CLR_TX_ABRT,1)
        status = mem32[ self.i2c_base | self.IC_CLR_RD_REQ]
        mem32[ self.i2c_base | self.IC_DATA_CMD] = data  & 0xff

        
        

    def any(self):
        # get IC_STATUS
        status = mem32[ self.i2c_base | self.IC_STATUS]
        # check RFNE receive fifio not empty
        if (status &  8) :
            return True
        return False
    
    def get(self):
        while not self.any():
            pass
        return mem32[ self.i2c_base | self.IC_DATA_CMD] & 0xff
    
    if __name__ == "__main__":
        import utime
        from machine import mem32
        from i2cSlave import i2c_slave
        
        s_i2c = i2c_slave(0,sda=0,scl=1,slaveAddress=0x41)
        counter =1
        try:
            while True:
                if s_i2c.any():
                    #print(s_i2c.get())
                    aa = s_i2c.get()
                    print(aa)
                    if aa == 12:
                        led_onboard.value(1)
                    if aa == 15:
                        led_onboard.value(0)
                if s_i2c.anyRead():
                    counter = counter + 1
                    s_i2c.put(counter & 0xff)
                    print(counter)
            
        except KeyboardInterrupt:
            pass
`
an error message with small window, with text inside "ManagementError" is appeared:
`STDERR:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 1, in <dictcomp>
NameError: name '__thonny_helper' isn't defined`

aivarannamaa, 2021-04-27:
Thank you for the report!

You found quite a rabbit hole!

When I tried to reproduce your case, I first got an error about missing module `i2cSlave` and then the part of error you posted. 
Are you sure you don't have more errors in Thonny's shell? Before STDERR?

Anyway, with my investigations I found that something gets broken in MicroPython VM when it meets an error on the class level (`from i2cSlave import` in my case). I forwarded this to MicroPython developers: https://github.com/micropython/micropython/issues/7171

Note that you have quite unusual usage of `if __name__ == "__main__"` -- usually it's on the module level but you have it at the class level.

California0306, 2021-04-27:
Like I said, only those two error messages appeared. I got errors also on second pico when doing experiments and this happend only when i2c was involved. Somehow situation is normalised now, but I don't know how and why. 
Regards, Vlado

closed 2021-06-13 by aivarannamaa

ghjwp7, 2024-12-28:
I realize this is a closed issue (as are some of the referenced issues, 1769, 1782, 1788, 2062) but I don't understand what the resolution is, ie how to get past the problem.  Here is the code which, when pasted into Shell, produces a "NameError: name '__thonny_helper' isn't defined" error that I wish to avoid:
```
import rp2
@rp2.asm_pio(set_init=rp2.PIO.OUT_LOW)
def pioProg0():
    nop()
```
In context, in freshly started thonny:
```
MicroPython v1.21.0 on 2023-10-06; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> import rp2
@rp2.asm_pio(set_init=rp2.PIO.OUT_LOW)
def pioProg0():
    nop()
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "rp2.py", line 262, in dec
  File "<stdin>", line 4, in pioProg0
NameError: name '__thonny_helper' isn't defined
>>> 
```

aivarannamaa, 2025-01-15:
@ghjwp7 , which Thonny version are you using?

ghjwp7, 2025-02-01:
> [@ghjwp7](https://github.com/ghjwp7) , which Thonny version are you using?

At ubuntu.22.04 linux command line:
> thonny --version
INFO    thonny: Thonny version: 3.3.14

In thonny after startup with an RP2040 board attached:
MicroPython v1.25.0-preview.160.gc73204128 on 2024-12-30; Raspberry Pi Pico with RP2040

(Sorry for delay, didn't see question until today 31 Jan)

aivarannamaa, 2025-02-04:
Thonny 3.3.14 is quite old. Please try 4.1.7.

ghjwp7, 2025-02-05:
> Thonny 3.3.14 is quite old. Please try 4.1.7.

Ok, I'll update.  It will be a couple of days since I'm installing a new computer.