thonny/thonny #1784: ERROR   thonny.plugins.micropython.backend: Crash in backend

matijaz74, 2021-04-27:
Hi!
The last few days I get an error that follows then.
It happens when I disconnect the Raspberry Pico USB cable, then reconnect and follow the instructions
`Backend terminated or disconnected. Use 'Stop/Restart' to restart.`

after pressing the stop button I get errors:

```
ERROR   thonny.plugins.micropython.backend: Crash in backend
Traceback (most recent call last):
  File "/home/matijaz/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 135, in __init__
    self._prepare_after_soft_reboot(clean)
  File "/home/matijaz/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 154, in _prepare_after_soft_reboot
    self._execute_without_output(script)
  File "/home/matijaz/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 533, in _execute_without_output
    raise ManagementError(script, out, err)
thonny.plugins.micropython.backend.ManagementError: Problem with a management command

SCRIPT:

class __thonny_helper:
    try:
        import uos as os
    except ImportError:
        import os
    import sys

    # for object inspector
    inspector_values = dict()
    last_repl_values = []
    @classmethod
    def print_repl_value(cls, obj):
        if obj is not None:
            cls.last_repl_values.append(obj)
            cls.last_repl_values = cls.last_repl_values[-5:]
            print('[object_link_for_thonny=%d]' % id(obj), repr(obj), '', sep='')

    @staticmethod
    def print_mgmt_value(obj):
        print('<thonny>', repr(obj), '</thonny>', sep='', end='')

    @staticmethod
    def repr(obj):
        try:
            s = repr(obj)
            if len(s) > 50:
                s = s[:50] + "..."
            return s
        except Exception as e:
            return "<could not serialize: " + str(e) + ">"

    @classmethod
    def listdir(cls, x):
        if hasattr(cls.os, "listdir"):
            return cls.os.listdir(x)
        else:
            return [rec[0] for rec in cls.os.ilistdir(x) if rec[0] not in ('.', '..')]

    @classmethod
    def getcwd(cls):
        if hasattr(cls, "getcwd"):
            return cls.os.getcwd()
        else:
            # micro:bit
            return ""

    @classmethod
    def chdir(cls, x):
        return cls.os.chdir(x)

    @classmethod
    def rmdir(cls, x):
        return cls.os.rmdir(x)

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 6
IndentationError: unindent doesn't match any outer indent level

Backend terminated or disconnected. Use 'Stop/Restart' to restart.
```

What do I have to do to prevent this from happening? I use Thonny 3.3.6

Thanks for help and answare!

aivarannamaa, 2021-04-28:
Thank you for the report!

Are you using threads in your Pico code?

matijaz74, 2021-04-28:
A few days ago, I wrote a code for a traffic light where threads from the “Get started with MicroPython on Raspberry Pi Pico” learning material are used. An error occurred and I didn’t have to connect to Thonny until I mechanically reset it. I noticed later that there was an error already in this learning material that spoiled the operation of the Raspberry Pi Pico microcontroller.
This error still occurs if I turn off the Raspberry Pi Pico and then plug it back in, and click Stop / Restart. The second time I click, the connection is set up normally.

aivarannamaa, 2021-04-28:
From your error message I read that Thonny sent some code to MicroPython for define a helper object for itself, but MicroPython received this with a syntax error. This means the data got corrupted during transfer. I've seen it happening when the background thread is still running (https://github.com/micropython/micropython/issues/6899), but apparently there can be other reasons for this.

Does this problem occur also after hard reset? If yes, then it would be of great help if you could progressively simplify your main.py and see what makes this problem disappear.

matijaz74, 2021-04-28:
I have to thank you very much. Now everything is as it was in the beginning. But I didn't think of that. Thank you for your help. I knew where to turn. I deleted main.py from Pico and the error no longer occurred. I apparently the main.py file went bad. Thanks!

closed 2021-06-13 by aivarannamaa