thonny/thonny #1792: No PWM output on ESP32 when running a script after the first time.

clrdls, 2021-04-30:
Using Thonny 3.3.7 with an ESP32 using WebREPL with a micropython script to ramp up the output of an LED, the LED ramps up as expected the first time the script is run after a power up or reset but if the script is stopped and run again the program shows the LED duty cycle increasing but there is no output on the pin. The LED output works as expected when connected by USB.  This is true on at least 5 pins that I have tried. The script is attached.
[test2.txt](https://github.com/thonny/thonny/files/6408420/test2.txt)

aivarannamaa, 2021-05-01:
Have you tried it outside of Thonny? Does it work when you store the script as main.py and soft-reboot the device?

clrdls, 2021-05-02:
PWM does not work properly using WebREPL through webrepl.html.  Also doing a soft reboot does not make it work either.  I found on github micropython/micropython issue 5737 that does work.   After creating the PWM object from a pin, deinitialize the object then initialize it in the script before using it.  
https://github.com/micropython/micropython/issues/5737

aivarannamaa, 2021-05-02:
Do I understand correctly that the problem was not related to Thonny?

clrdls, 2021-05-02:
That is right, it seems to be a WebREPL problem.

closed 2021-05-03 by aivarannamaa