thonny/thonny #1795: backend issue

WDC56, 2021-05-02:
canTraceback (most recent call last):
  File "C:\Users\dylan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 511, in _start_runner
    self._runner.start()
  File "C:\Users\dylan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 116, in start
    self.restart_backend(False, True)
  File "C:\Users\dylan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 602, in restart_backend
    self._proxy = backend_class(clean)
  File "C:\Users\dylan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\__init__.py", line 270, in __init__
    super().__init__(clean, get_interpreter_for_subprocess(executable))
  File "C:\Users\dylan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\__init__.py", line 38, in __init__
    super().__init__(clean, executable)
  File "C:\Users\dylan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 866, in __init__
    self._start_background_process(clean=clean)
  File "C:\Users\dylan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 933, in _start_background_process
    **extra_params
  File "C:\Users\dylan\AppData\Local\Programs\Thonny\lib\subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "C:\Users\dylan\AppData\Local\Programs\Thonny\lib\subprocess.py", line 1207, in _execute_child
    startupinfo)
PermissionError: [WinError 5] Access is denied

t use suddenly ,was ok ,now showing

aivarannamaa, 2021-05-02:
Are you using the interpreter that comes with Thonny or a separate Python? If you don't know, then it's probably the former.

Please try following:

* Select "Tools => Open Thonny data folder"
* Close Thonny
* In the data folder make copy of configuration.ini and delete the original
* Start Thonny again

If this helps, then please send me the copy of configuration file.

WDC56, 2021-05-08:
I removed all files to do with thonny ,then uninstalled then reinstalled ,and all seems to be good now 
              thank you

closed 2021-05-08 by WDC56