thonny/thonny #1800: Crash at start : "BadLength (poly request too large or internal Xlib length error)"

JerardMarca, 2021-05-02:
hello everyone,

I'm using Thonny 3.3.5 for a few weeks, and never got any errors. I'm using Linux Mint 19.3 Xfce 32bits. Today, I was editing some code with discord.py module, and it needed to connect to the bot on the site (no problems here, this is just the context). I closed Thonny with all my bots stopped, and then when I tried to re-open it using my normal thonny.desktop file from the applications menu, it appears at the screen, and closes just after. My task manager shows that effectively Thonny is running, and just after it stops itself.

Here is the result when I launch the thonny.sh file from the terminal : 

`INFO    thonny: Thonny version: 3.3.5`
`X Error of failed request:  BadLength (poly request too large or internal Xlib length error)`
`Major opcode of failed request:  139 (RENDER)`
`Minor opcode of failed request:  20 (RenderAddGlyphs)`
`Serial number of failed request:  3813`
`Current serial number in output stream:  3814`

Any idea to how I can resolve this ?
I tried restarting my PC, but nothing changes.

aivarannamaa, 2021-05-02:
Thank you for the report!

Unfortunately this error message doesn't ring the bell for me. Please try the following:

* Find Thonny's configuration folder. By default it's `~/.config/Thonny`
* Rename `configuration.ini` to something else
* Try to start Thonny again

If this helps, then please send me the original configuration file (which was causing problems).

JerardMarca, 2021-05-03:
Hello,

Thanks for the answer ! Effectively, it worked !
As I can see, I did what you say, and it created another configuration.ini in `~.config/Thonny` file when I launched again Thonny, and then it worked perfectly (had to re do my settings, but no problems here) !

I send you the original configuration file which was causing problems (it is in .txt cuz GitHub does not support .ini files)
[configuration.ini.txt](https://github.com/thonny/thonny/files/6413549/configuration.ini.txt)
.

Thanks for your help !

aivarannamaa, 2021-05-03:
Thank you for the configuration.ini!

Unfortunately I'm as puzzled as before. I expected to see some extra large or extra small dimensions, but everything looks normal.

aivarannamaa, 2021-05-03:
Were you using emoji or some other unusual characters in your scripts?

JerardMarca, 2021-05-04:
@aivarannamaa No, I'm not using special characters or anything, I write in english, using only the letters of my keyboard (if you wonder why they are "*" symbols in the .ini file, this is just me hiding my name, otherwise this is normal english ASCII).

aivarannamaa, 2021-05-04:
I'm closing this for now. Please reopen if this happens again.

closed 2021-05-04 by aivarannamaa