thonny/thonny #1808: SSH Port Adreess

dk2jk, 2021-05-06:


aivarannamaa, 2021-05-06:
Unfortunately, the port is not configurable at the moment. I'll add the possibility in the next version.

Until then your only option is to modify Thonny's souce code. Select "Tools => Open Thonny program folder", find this location: https://github.com/thonny/thonny/blob/40fbb4bd97a5dfb9d008e0519b73c7e9b85e0b4c/thonny/backend.py#L523 and add `port=...` parameter.

dk2jk, 2021-05-06:
tnx , works

ansobczak, 2022-02-24:
thanks, helps a lot!

vinniec, 2023-08-21:
I can't connect even applying the code modification that is suggested here.
```
Can't connect to '10.0.0.2' with user 'u0_a281': Authentication failed.
Re-check your host, authentication method, password or keys.

Process ended with exit code 1.
```
However I realized that I can write the port in the Host field, for example in my case 10.0.0.2:8022
If it helps.

closed 2024-12-28 by aivarannamaa

Mr-bkmtech, 2026-11-14:
Fix this fucking problem.   My work is being held up over a 5-year-old issue.