thonny/thonny #1817: NRF52840 is very hard to use the ctrl+c to stop the upload program

genvex, 2021-05-13:
I thonny to work with the NRF52840 ,I like the rich view pad thonny has, but after I upload the program, it is hard for me to stop and continue to make code, but the Mu works well

aivarannamaa, 2021-05-13:
Thank you for the report!

Could you please give more details? Why is it hard to use Ctrl+C?

toybuilder, 2021-05-19:
This can sometimes happen if one has code running in interrupt callbacks that prints stuff out.  I just recently made a mistake of having my fast-running timer-tick callbacks print out a debugging string.

genvex, 2021-05-19:
But the MU work so well, althought I love thonny more. 

　　　　　　　　　
 
From: Joseph Chiu
Date: 2021-05-19 10:56
To: thonny/thonny
CC: genvex; Author
Subject: Re: [thonny/thonny] NRF52840 is very hard to use the ctrl+c to stop the upload program (#1817)
This can sometimes happen if one has code running in interrupt callbacks that prints stuff out. I just recently made a mistake of having my fast-running timer-tick callbacks print out a debugging string.

aivarannamaa, 2021-05-19:
@genvex, can you give me example code and instructions for reproducing your problem? Which board and which MicroPython or CircuitPython version are you using? Also which Thonny version and which OS?

aivarannamaa, 2021-06-13:
@genvex , please reopen if you have more details to add!

closed 2021-06-13 by aivarannamaa