thonny/thonny #1818: debug mode gives "ERROR thonny.backend: ... AttributeError: module 'numbers' has no attribute 'Number'"

dxxxm, 2021-05-14:
any attempt to debug ends with `AttributeError: module 'numbers' has no attribute 'Number'`
```
Python 3.9.2 (/home/mario/workspace/Python/Thonny/.py3/bin/python)
>>> %Debug ss.py
1
3
2
4
3
5
42
44
i: 42
ERROR   thonny.backend: PROBLEM WITH THONNY'S BACK-END
Traceback (most recent call last):
  File "/home/mario/workspace/Python/Thonny/.py3/lib/python3.9/site-packages/thonny/plugins/cpython/cpython_backend.py", line 1278, in execute_source
    root = self._prepare_ast(source, filename, mode)
  File "/home/mario/workspace/Python/Thonny/.py3/lib/python3.9/site-packages/thonny/plugins/cpython/cpython_backend.py", line 1736, in _prepare_ast
    ast_utils.mark_text_ranges(root, source)
  File "/home/mario/workspace/Python/Thonny/.py3/lib/python3.9/site-packages/thonny/ast_utils.py", line 183, in mark_text_ranges
    ASTTokens(source, tree=node)
  File "/home/mario/workspace/Python/Thonny/.py3/lib/python3.9/site-packages/asttokens/asttokens.py", line 65, in __init__
    self.mark_tokens(self._tree)
  File "/home/mario/workspace/Python/Thonny/.py3/lib/python3.9/site-packages/asttokens/asttokens.py", line 76, in mark_tokens
    MarkTokens(self).visit_tree(root_node)
  File "/home/mario/workspace/Python/Thonny/.py3/lib/python3.9/site-packages/asttokens/mark_tokens.py", line 49, in visit_tree
    util.visit_tree(node, self._visit_before_children, self._visit_after_children)
  File "/home/mario/workspace/Python/Thonny/.py3/lib/python3.9/site-packages/asttokens/util.py", line 204, in visit_tree
    ret = postvisit(current, par_value, value)
  File "/home/mario/workspace/Python/Thonny/.py3/lib/python3.9/site-packages/asttokens/mark_tokens.py", line 92, in _visit_after_children
    nfirst, nlast = self._methods.get(self, node.__class__)(node, first, last)
  File "/home/mario/workspace/Python/Thonny/.py3/lib/python3.9/site-packages/asttokens/mark_tokens.py", line 319, in visit_const
    if isinstance(node.value, numbers.Number):
AttributeError: module 'numbers' has no attribute 'Number'
>>> 
```
newly created virtual env
```
(.py3) mario@wally:~/workspace/Python/Thonny$ python -V
Python 3.9.2            
(.py3) mario@wally:~/workspace/Python/Thonny$ lsb_release -a                                                                                                        
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux bullseye/sid
Release:        testing
Codename:       bullseye

(.py3) mario@wally:~/workspace/Python/Thonny$ pip list -l 
Package           Version
----------------- --------
astroid           2.5.6
asttokens         2.0.5
docutils          0.17.1
isort             5.8.0
jedi              0.18.0
lazy-object-proxy 1.6.0
mccabe            0.6.1
mypy              0.812
mypy-extensions   0.4.3
parso             0.8.2
pip               21.1.1
pkg-resources     0.0.0
pylint            2.8.2
pyserial          3.5
Send2Trash        1.5.0
setuptools        56.2.0
six               1.16.0
thonny            3.3.7
toml              0.10.2
typed-ast         1.4.3
typing-extensions 3.10.0.0
wrapt             1.12.1
```

aivarannamaa, 2021-05-14:
Thank you for the report!

Can you give me the program that causes this?

aivarannamaa, 2021-05-14:
... also, do you happen to have a script named `numbers.py` or directory named `numbers` in your working directory?

dxxxm, 2021-05-15:
Good point. 
Directory contained a file `numbers.py`. Renaming it solved it. All good.
Sorry for the false alarm.

closed 2021-05-15 by dxxxm