thonny/thonny #1820: Over-the-air feature

phme283, 2021-05-17:
Is it planned to add an over-the-air feature? So that you are able to flash your code via Wifi?

aivarannamaa, 2021-05-17:
I assume you mean Thonny's MicroPython support and the ability to update py files on the device.

Updating files over WiFi is already possible if your device supports WebREPL.

Which device are you using?

phme283, 2021-05-17:
@aivarannamaa: Thanks a lot for you quick answer!
Currently, I am starting with a first small home IoT project using an ESP32 and a Raspberry Pi as "server". 

I was not aware of WebREPL and will have a deeper look into it. Google always brought me to "over the air" with the Arduino IDE. So, thanks a lot for your helpful answer.

closed 2021-05-17 by phme283