thonny/thonny #1821: thonny won't detect my R.S Pico/

SamTrick, 2021-05-17:


aivarannamaa, 2021-05-17:
Need more details. Did you try selecting a specific port? Which platform are you using?

micahsgilbert, 2021-05-19:
I'm having the same issue so maybe what I'm having is the same as who opened this issue. I'm on Ubuntu 20.04 and Thonny only has `Try to detect port automatically` in the options to select the port. I know that the USB cable I'm using is fine and the Pico is working because I can upload the CircuitPython firmware just fine. However, when I try to connect it normally, I get nothing added to the output of `lsusb` and nothing like `/dev/ttyACM0` is created.

aivarannamaa, 2021-06-13:
@micahsgilbert, if the port doesn't appear, then it's outside of Thonny's scope. I suggest you ask about this in the MicroPython section of Raspberry Pi forum.

closed 2021-06-14 by aivarannamaa