thonny/thonny #1824: Icons not fitting

speedy-10, 2021-05-19:
Hi, using Thonny 3.3.10 on ubuntu 20.04 with a 4k-monitor the icons don't really fit in my opinion, see picture. Furthermor in the first menu > "Move/rename" isn't translated in German so far.
[image]
Regards

aivarannamaa, 2021-06-12:
Thank you for the report and sorry for the late reply! I somehow missed the notification.

"Move / rename" will be available for translation when I'm ready to launch first betas of Thonny 4.0.

I'll see what I can do about the icons. I don't have a high-res monitor, so it's not easy to reproduce this.

speedy-10, 2021-06-12:
Hi, if I can help you, I could install a new version like you wrote me some time ago:
`python3 -m venv thonny_venv
source thonny_venv/bin/activate
pip install --pre --no-cache-dir -U thonny`
... test it and give you feedback, if the issue is done.
Regards