thonny/thonny #1825: Manage Packages results in error for Adafruit-Blinka

makermelissa, 2021-05-19:
When I try and install Adafruit-Blinka from PyPI, I get an error:
[image]

aivarannamaa, 2021-05-19:
Thank you for the report!

Please try the latest release: https://github.com/thonny/thonny/releases/tag/v3.3.10

Out of curiosity, which device are you using?

makermelissa, 2021-05-19:
Ok, I will. Raspberry Pi Pico with MicroPython. Manually downloading and copying works fine.

makermelissa, 2021-05-19:
Thanks, it appears to fix this error. Another error occurs, but I suspect it's installing linux packages and running out of room.

aivarannamaa, 2021-05-20:
I now discovered, that 64-bit Linux bundle of Thonny 3.3.10 has problems installing with pip: `ImportError: libffi.so.6: cannot open shared object file: No such file or directory`. I suppose it is problem with bundling.

When I try to install Adafruit-blinka with Thonny in my dev-environment, Pico runs out of flash. If there was an easy way to remove irrelevant platforms from blinka, I could add this to [minipip](https://github.com/aivarannamaa/minipip/), which powers Thonny's MicroPython package management since 3.3.10.

I read your post about Blinka (https://learn.adafruit.com/circuitpython-libraries-on-micropython-using-the-raspberry-pi-pico/installing-blinka-and-libraries), so I assume you're interested in making package management smoother for CircuitPython and MicroPython. So am I, that's why I started minipip. If you want to know my ideas and visions, then I'm happy to share.

makermelissa, 2021-05-20:
Thanks for chiming in @aivarannamaa. I was going to contact you later regarding this anyways, so this is perfect. I'm definitely interested in making the install smoother. Most of the files in requirements are not needed. Blinka is intended to run on either MicroPython or CPython which is the reason for the dependencies. I'm really seeing 3 issues that if fixed would result in a perfect minimal install. 

1. Let's start with the dependencies. Basically, the only requirement needed for MicroPython is Adafruit-Python-PlatformDetect. Everything else can be ignored. I'm not sure what would work the best while still maintaining CPython compatibility.

2. The next issue is the root folder. For Blinka, the relevant files are inside of the src folder. For PlatformDetect, the only file copied over should be the adafruit_platformdetect. I noticed the /bin folder was being copied over.

3. Finally other issue is that most of the boards and microcontrollers in Blinka (under src/adafruit_blinka) are not needed for MicroPython. Here's a list of the folders that are MP related:
 - src/adafruit_blinka/board/raspberrypi/pico.py
 - src/adafruit_blinka/board/pyboard.py
 - src/adafruit_blinka/microcontroller/generic_micropython
 - src/adafruit_blinka/microcontroller/esp8266
 - src/adafruit_blinka/microcontroller/rp2040
 - src/adafruit_blinka/microcontroller/stm32/stm32f405
Everything else is linux. I'm not sure what would be the easiest way to maintain a list in case we want to add more MP boards in the future. Here's an issue on the Adafruit side that has a minimal Raspberry Pi Pico version running: https://github.com/adafruit/Adafruit_Blinka/issues/472

aivarannamaa, 2021-05-20:
1. I think different deps could be marked with different *environment markers* (https://www.python.org/dev/peps/pep-0496/ and https://www.python.org/dev/peps/pep-0508/#environment-markers) and minipip could pass suitable `--platform` to pip (https://pip.pypa.io/en/stable/cli/pip_install/#cmdoption-platform). I haven't tried it, but it may work.
2. I forgot about bin directories. minipip could use some heuristics to get rid of unwanted directories.
3. We could agree on a metadata file, which, if present, lists the files that are relevant for MicroPython. Alternatively (or in addition to this) you could teach Adafruit-Python-PlatformDetect to post-process the pip-installed files, so that minipip could ask it to remove irrelevant files before concluding its job.

makermelissa, 2021-05-21:
Ok yeah, I'll have to look into wheels.

aivarannamaa, 2021-05-21:
I also got curious and made an experiment: https://pypi.org/project/mp-packaging-experiment/#files

Here I created several wheels with `micropython` Python tag (https://www.python.org/dev/peps/pep-0425/#python-tag).

I can download and extract any of them with command like this:

```
pip3 install --target=temp_dir mp_packaging_experiment --implementation=micropython --python-version=1 --only-binary=:all:
```
Note that `micropython1` gets split to two arguments: `--implementation=micropython` and `--python-version=1` (https://discuss.python.org/t/how-does-implementation-option-of-pip-install-work/8830).

The argument `--only-binary=:all:` is required by `pip`, whenever `--implementation` or `--python-version` is used. This means all dependencies must also have wheels with same Python version tag.

I imagine the more general tag `micropython1` could be used for uncompiled packages (ie containing py-scripts) and `micropython115` etc could be used for compiled packages (ie. containing mpy files).

I didn't find good means for specifying arbitrary python tag while building the wheel (I guess you are supposed to build a wheel on the same platform the wheel is for). So I created a regular py3 wheel, unpacked it, made changes in the content and metadata and then repacked it: https://github.com/aivarannamaa/packaging_experiment/blob/master/create_dist.sh

makermelissa, 2021-05-21:
Fortunately for MP, all files Blinka are python and not binary.

makermelissa, 2021-05-21:
Tagging @ladyada...

aivarannamaa, 2021-05-22:
@makermelissa and @ladyada, I suggest we continue this discussion under https://github.com/aivarannamaa/minipip/issues/5, as Thonny will inherit all the advances in `minipip`.

In fact, all current issues under https://github.com/aivarannamaa/minipip/issues might be interesting for you.

closed 2021-06-14 by aivarannamaa