thonny/thonny #1827: Pip defaulting to sdist over bdist

schellenberg, 2021-05-20:
When attempting to install PyQt5 on Windows 10 via the graphical pip installer, I get the following error:

`Collecting PyQt5
  Downloading PyQt5-5.15.4-cp36.cp37.cp38.cp39-none-win32.whl (5.4 MB)
Collecting PyQt5-Qt5>=5.15
  Downloading PyQt5_Qt5-5.15.2-py3-none-win32.whl (43.4 MB)
Collecting PyQt5-sip<13,>=12.8
  Downloading PyQt5_sip-12.9.0.tar.gz (103 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
    Preparing wheel metadata: started
    Preparing wheel metadata: finished with status 'done'
Building wheels for collected packages: PyQt5-sip
  Building wheel for PyQt5-sip (PEP 517): started
  Building wheel for PyQt5-sip (PEP 517): finished with status 'error'
  ERROR: Command errored out with exit status 1:
   command: 'C:\Users\schel\AppData\Local\Programs\Thonny\python.exe' 'C:\Users\schel\AppData\Roaming\Python\Python37\site-packages\pip\_vendor\pep517\in_process\_in_process.py' build_wheel 'C:\Users\schel\AppData\Local\Temp\tmp2197bhxz'
       cwd: C:\Users\schel\AppData\Local\Temp\pip-install-gexpgc5c\pyqt5-sip_7e2acf04c20f40de8ce666bc23faa0f3
  Complete output (5 lines):
  running bdist_wheel
  running build
  running build_ext
  building 'PyQt5.sip' extension
  error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
  ----------------------------------------
  ERROR: Failed building wheel for PyQt5-sip
Failed to build PyQt5-sip
ERROR: Could not build wheels for PyQt5-sip which use PEP 517 and cannot be installed directly
Process returned with code 1`

As you can see in the error message, pip is opting for the sdist tarball for the PyQt5-sip package. However, if I use Tools --> Open system shell... and type in
``pip install --only-binary :all: PyQt5`` 
the install works successfully. 

Is there some way to have the graphical installer default to using the built bdist when available? This would help my students immensely when they are trying to install packages.

Ivoz, 2021-05-20:
It is possible that maybe this is caused by the latest 12.9 PyQt5-sip wheels only being available for [win64](https://pypi.org/project/PyQt5-sip/#files] whereas the previous release 12.8.1 had win32 available.

I think by default thonny bundles a 32 bit python.

Therefore there are fixes/workarounds:
* Use a thonny with an inbuilt x64 python
* Configure thonny to use an x64 python binary
* Install PyQt5-sip 12.8.1 first
* Ask the PyQt5-sip maintainer (riverbankcomputing?) if they could kindly still release 32 bit wheels

This may also be an argument for thonny providing a fully x64-by-default version in the future

aivarannamaa, 2021-05-20:
That's correct, Thonny's Windows bundle comes with 32-bit Python. Ivoz is also correct about possibility to configure Thonny to use an separate x64 Python installation.

Starting with Thonny 4.0, I'll publish both 64-bit and 32-bit bundles for Windows. I already experimented with this: https://github.com/thonny/thonny/releases/download/v3.3.7/thonny-3.3.7-alt.exe comes with 64-bit Python 3.9

schellenberg, 2021-05-21:
Thanks! You are definitely correct that this was entirely due to the lack of a 32-bit wheel in the dependency. I appreciate your help.

closed 2021-05-21 by schellenberg