thonny/thonny #1843: Flask "app.run(debug=True)" unable to execute

smfrederico, 2021-05-26:
I ran this "main.py" to Implement the Flask Class whose last line was "app.run(debug=True)" in Thonny . I got this message error "%Run main.py

* Serving Flask app "main" (lazy loading)

* Environment: production

   WARNING: This is a development server. Do not use it in a production deployment.

   Use a production WSGI server instead.

* Debug mode: on

* Restarting with stat

C:\Users\Frederico\AppData\Local\Programs\Thonny\python.exe: No module named thonny.plugins.cpython.main ".

I only solved it when changing to "app.run(debug=False)". I was wondering if I should have installed another library, such as thonny.plugins, but I didn´t manage to find such one. Would anyone
have any idea why I got this warning? Thanks in advance.

Below I will paste the main.py:
from flask.views import MethodView
from wtforms import Form,StringField,SubmitField
from flask import Flask,render_template,request
from calorie import Calorie
from temperature import Temperature

app = Flask(__name__)

class HomePage(MethodView):
    def get(self):
        return render_template('index.html')
class CaloriesFormPage(MethodView):
    def get(self):
        calories_form = CaloriesForm()
        return render_template('calories_form_page.html',caloriesform=calories_form)
    def post(self):
        calories_form = CaloriesForm(request.form)
        temperature = Temperature(country=calories_form.country.data,city=calories_form.city.data).get()
        calorie = Calorie(weight=float(calories_form.weight.data),height=float(calories_form.height.data),age=float(calories_form.age.data),temperature=temperature)
        calories = calorie.calculate()
        return render_template('calories_form_page.html',caloriesform=calories_form,calories=calories,result=True)
class CaloriesForm(Form):
    weight = StringField("Weight: ", default=70)
    height = StringField("Height: ", default=175)
    age = StringField("Age: ",default=32)
    country = StringField("Country: ",default='USA')
    city = StringField("City: ", default='San Francisco')
    button = SubmitField("Calculate")
    
app.add_url_rule('/', view_func=HomePage.as_view('home_page'))
app.add_url_rule('/calories_form',view_func=CaloriesFormPage.as_view('calories_form_page'))
app.run(debug=False)

aivarannamaa, 2021-06-12:
Thank you for the report and sorry for late reply! I somehow missed the notification.

I'll deal with this issue under #1781

closed 2021-06-12 by aivarannamaa

dss010101, 2023-04-16:
this still seems to be happening. how do you solve this?

aivarannamaa, 2023-04-16:
@dss010101, please open new issue and describe, what is still happening.