thonny/thonny #1845: Big Sur 11.4 Thonny:  3.7.9 or 3.10.3 Numpy listed in Manage Plug-ins but import fails

cnoblejr, 2021-05-26:
Traceback (most recent call last):
  File "/Users/carlnoble27/IDrive-Sync/TurtleRace01.py/IDrive-Sync/t.py", line 11, in <module>
    import numpy
  File "/Users/carlnoble27/Library/Python/3.7/lib/python/site-packages/numpy/__init__.py", line 148, in <module>
    from . import lib
  File "/Users/carlnoble27/Library/Python/3.7/lib/python/site-packages/numpy/lib/__init__.py", line 37, in <module>
    from .npyio import *
  File "/Users/carlnoble27/Library/Python/3.7/lib/python/site-packages/numpy/lib/npyio.py", line 14, in <module>
    from ._datasource import DataSource
  File "/Users/carlnoble27/Library/Python/3.7/lib/python/site-packages/numpy/lib/_datasource.py", line 38, in <module>
    import shutil
  File "/Users/carlnoble27/IDrive-Sync/TurtleRace01.py/IDrive-Sync/shutil.py", line 19, in <module>
    from . import tarfile
ImportError: attempted relative import with no known parent package

cnoblejr, 2021-05-26:
Problem when saving to iDrive-Sync.  Sving to Desktop or Documents works fine.  Have no idea why.

aivarannamaa, 2021-05-27:
In Python by default the current directory is the primary candidate for locating modules. In your case `numpy` wanted to import `shutil` from the standard library, but the `shutil.py` in your `IDrive-Sync` folder was shadowing the right module, so `numpy` imported the wrong module.

I don't know how IDrive-Sync works and why there is so strange path mentioned in the stacktrace: `/Users/carlnoble27/IDrive-Sync/TurtleRace01.py/IDrive-Sync/shutil.py`. Is this `shutil.py` required in IDrive-Sync folder?

closed 2021-06-14 by aivarannamaa