thonny/thonny #1846: Text animation in shell does not work as well in newer versions of Thonny

Skparab1, 2021-05-29:
I've been using Thonny for a while and text animation used to work very well in earlier versions of Thonny (like Thonny 3.3.0), but when I updated to the latest version of Thonny 3.3.10, text animations blink a lot. An example of this is my game flappy bird (https://github.com/Skparab1/flappy-bird). In earlier versions of Thonny, it used to work, great, but now it blinks a lot. I don't know what's causing that and whether it can be fixed or not, and it is not a big deal, but it would be great if you could try to fix it!
Thanks!
-Shubham

aivarannamaa, 2021-05-29:
Thank you for the report!

You make Thonny run the flappy bird program in Terminal, not in Thonny's Shell, right? Which OS and which terminal program are you using?

Skparab1, 2021-05-29:
Hi!
I actually make the program run in shell, I'll definitely try in terminal!
I'm using Mac os Calatina 10.15.7.
Thanks!

aivarannamaa, 2021-06-01:
I think the games like this are better to be run in Terminal. Thonny's shell is not optimized for this and I'm really surprised that your was playable in older Thonny versions.

When targeting terminal, you'll get to use full range of ANSI codes for managing the screen, eg. `print("\033[2J")` for clearing the screen. See https://en.wikipedia.org/wiki/ANSI_escape_code#CSI_(Control_Sequence_Introducer)_sequences for an overview of the options.

closed 2021-06-01 by aivarannamaa