thonny/thonny #1851: Backend terminated or disconnected.Windows fatal exception: access violation

alfisyah98, 2021-06-01:
please help me.
Thread 0x00001d04 (most recent call first):
  File "D:\PERKULIAHAN\TA VERSI NEW\data training\eigen3.py", line 55 in <module>
  File "c:\users\user\appdata\local\programs\python\python38\lib\site-packages\thonny\backend.py", line 1272 in _execute_prepared_user_code
  File "c:\users\user\appdata\local\programs\python\python38\lib\site-packages\thonny\backend.py", line 1200 in wrapper
  File "c:\users\user\appdata\local\programs\python\python38\lib\site-packages\thonny\backend.py", line 1213 in wrapper
  File "c:\users\user\appdata\local\programs\python\python38\lib\site-packages\thonny\backend.py", line 1259 in execute_source
  File "c:\users\user\appdata\local\programs\python\python38\lib\site-packages\thonny\backend.py", line 814 in _execute_source
  File "c:\users\user\appdata\local\programs\python\python38\lib\site-packages\thonny\backend.py", line 800 in _execute_file
  File "c:\users\user\appdata\local\programs\python\python38\lib\site-packages\thonny\backend.py", line 403 in _cmd_Run
  File "c:\users\user\appdata\local\programs\python\python38\lib\site-packages\thonny\backend.py", line 204 in handle_command
  File "c:\users\usera\appdata\local\programs\python\python38\lib\site-packages\thonny\backend.py", line 146 in mainloop
  File "c:\users\user\appdata\local\programs\python\python38\lib\site-packages\thonny\backend_launcher.py", line 87 in <module> Use 'Stop/Restart' to restart.

aivarannamaa, 2021-06-01:
Can you post the program you were trying to run?

alfisyah98, 2021-06-01:
from time import time
import numpy, os
import matplotlib.pyplot as plt
import logging
from sklearn.metrics import classification_report
from sklearn.metrics import confusion_matrix
from sklearn.metrics import accuracy_score
from sklearn.model_selection import train_test_split
from sklearn.model_selection import GridSearchCV
from sklearn.decomposition import PCA
from sklearn.svm import SVC
import cv2
import pickle
from collections import Counter
  
def most_frequent(List):
    occurence_count = Counter(List)
    return occurence_count.most_common(1)[0][0]

pickle_in = open("pca3.dat","rb")
pca = pickle.load(pickle_in)

#X_train_pca = pca.transform(X_train)
#X_test_pca = pca.transform(X_test)

pickle_in = open("clf3.dat","rb")
clf= pickle.load(pickle_in)

#nama = {0: 'alfi', 1: 'falah', 2: 'fatan'}
nama = {0: 'alfi', 1: 'fatan', 2:'unknown'}

#proses face recogn
cap = cv2.VideoCapture(0, cv2.CAP_DSHOW)
face_cascade = cv2.CascadeClassifier('haarcascade_frontalface_default.xml')

cap.set(3,640)
cap.set(4,480)

identitas=[]
ketemu = False
#360x440
while(True):
    # Capture frame-by-frame
    test = []
    face = []
    ret, frame = cap.read()
    xv, yv, cv = frame.shape
    if ret == True :
        
        gray = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
        faces = face_cascade.detectMultiScale(gray, 1.3, 5)
        test = []
        for (x,y,wf,hf) in faces:
            # create box on detected face
            frame = cv2.rectangle(frame,(x,y),(x+wf,y+hf),(255,0,0),1)
            
            wajah = frame[y:y+hf,x:x+wf]
            dim = (320, 320)
  
            # resize image
            resized = cv2.resize(wajah, dim, interpolation = cv2.INTER_AREA)
            
            
            resized=cv2.cvtColor(resized, cv2.COLOR_BGR2GRAY)
            testImageFeatureVector=numpy.array(resized).flatten()
            test.append(testImageFeatureVector)
            testImagePCA = pca.transform(test)
            testImagePredict=clf.predict(testImagePCA)
            
            cv2.imshow('wframe', resized)
            print('ada wajah', nama[testImagePredict[0]], type(testImagePredict))
            print('.',end='')
            
            identitas.append(testImagePredict[0])
            if len(identitas) >= 30 :
                ketemu = True
                break
            
        cv2.imshow('frame', frame)
        
        if ketemu == True :
            break
        
        if cv2.waitKey(1) & 0xFF == ord('q'):
            break

# When everything done, release the capture
cap.release()
cv2.destroyAllWindows()

#print (identitas)
id_ = most_frequent(identitas)
print('ada wajah', nama[id_])

alfisyah98, 2021-06-01:
im confused. before install visual studios this programs is work but after install that, i have this message when im run

aivarannamaa, 2021-06-01:
Is it exactly the same program that caused the error?

The stacktrace says `File "D:\PERKULIAHAN\TA VERSI NEW\data training\eigen3.py", line 55 in`, but line 55 in this program is `test = []` -- I find it hard to believe this is causing the crash.

alfisyah98, 2021-06-01:
not line 55, but the error message in line 54 "faces = face_cascade.detectMultiScale(gray, 1.3, 5)"

alfisyah98, 2021-06-01:
do i reinstall this ide?

alfisyah98, 2021-06-01:
only in this windows, im trying on raspberry work.

aivarannamaa, 2021-06-01:
> `faces = face_cascade.detectMultiScale(gray, 1.3, 5)`

This could be an error in `cv2`. Can you try the same program outside of Thonny?

You could start by making Thonny run it in Terminal (Run => Run current script in Terminal). If it works, then the problem is in Thonny. If it doesn't work then try running it in VSCode. If it still doesn't work, then you should raise a CV2 issue.

alfisyah98, 2021-06-01:
> This could be an error in `cv2`. Can you try the same program outside of Thonny?
>
> You could start by making Thonny run it in Terminal (Run => Run current script in Terminal). If it works, then the [...]

not working when im running current script in terminal.. thank you for the response

alfisyah98, 2021-06-01:
for those of you who have the same problem, update your opencv

closed 2021-06-01 by alfisyah98