thonny/thonny #1854: Window resizing not working correctly with a rhs view open on 3.3.10 - macOS 11.4

marcoboerner, 2021-06-07:
When trying move the following edges or corners of the app window the app resizes in an unexpected way, as in the opposite sides start moving as well.

Issues with the following:
left edge
top left corner
bottom left corner
top right corner

One right hand side view (for example Variables) needs to be open to reproduce the issue.
Thonny 3.3.10
Raspberry Pi Dark UI Theme
Default Dark Syntax Theme

aivarannamaa, 2021-06-12:
Thank you for the report!

I've been told about such behavior before, but I haven't been able to reproduce it in Catalina.

Did you download Thonny from the homepage? If yes, then could you please try the alternative installer (https://github.com/thonny/thonny/releases/download/v3.3.10/thonny-3.3.10-alt.pkg), which has newer version of the UI toolkit.

aivarannamaa, 2021-06-12:
Oops, I was too quick. I tried only resizing from the left edge, but now I tried also left corners and the problem did manifest.

bhbr, 2025-04-24:
The problem persists on macOS Sonoma 14.6.1.
Resizing from the left edge or left corners moves the right edge to the right, either way the cursor in moved.
In addition, resizing from the top left corner moves both the right and bottom window edge in unpredictable ways.