thonny/thonny #1858: TypeError: a bytes-like object is required, not 'str'

DarkBasement, 2021-06-09:
I am working with Thonny 3.1.0 on raspberry PI. I debug a python script that uses flask. There are 'Code' and 'Shell' views on my screen.
When I set over app.run() I get an error window "Problem when handling 'ProgramOutput'". It looks like the web server is successfully running though.

In some cases I also get "TypeError: argument of type 'int' is not iterable". Just like #637 Wasn't it fixed?

Call stack:
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 1450, in event_generate
    handler(event)
  File "/usr/lib/python3/dist-packages/thonny/shell.py", line 225, in _handle_program_output
    self._insert_text_directly(msg.data, ("io", msg.stream_name))
  File "/usr/lib/python3/dist-packages/thonny/shell.py", line 422, in _insert_text_directly
    if "\a" in txt:
TypeError: a bytes-like object is required, not 'str'
```

aivarannamaa, 2021-06-12:
Thank you for the report!

I'm not familiar with the "Add / Remove software" window, but I suggest you execute following commands at Terminal:

```
sudo apt update
sudo apt upgrade
```

If this still doesn't give you Thonny 3.3.10, then you probably have the older Raspberry Pi OS (Stretch). See https://www.raspberrypi.org/documentation/raspbian/updating.md for upgrade instructions or backup your files and install Buster from scratch.

(Note that the next version of RPi OS (Bullseye) is coming this summer.)

aivarannamaa, 2021-07-06:
Please reopen if you still have this problem

closed 2021-07-06 by aivarannamaa