thonny/thonny #1859: Printing code (on paper) not consistant

Trinzini, 2021-06-10:
When printing out (on paper) the code for debugging etc. it prints usually in Black and white. Even when color is selected, Sometimes it prints some in color (variables etc) perhaps half way down the page then back to all black & white. Sometimes it uses color where it should correctly on the entire page. The first lines ( from machine import Pin) etc almost always are correct This makes for reading very annoying. I am using Windows 7 or Linux Mint/Manjaro and have the same problem. It is not the printer HP CP1025nw as it prints from Visual Basic and other apps perfectly. On the page preview you can see where it prints color correctly and where it's just black & white. Also when selecting the printer it shows a second copy of the printer. In control panel there is only 1 copy. Perhaps this issue can be addressed. Thank you.

aivarannamaa, 2021-06-10:
Thank you for the report!

Selecting printers is outside of Thonny's control -- this is provided by your browser.

Regarding wrong colors -- could you please post here an example html file having this problem? When you select "File => Print" then Thonny generates a html file into a temp folder and then asks your webbrowser to open it. The file includes a piece of JavaScript, which causes print dialog to open automatically. When you close this dialog, you should be able to right-click the html file and select view source.

aivarannamaa, 2021-07-06:
Please reopen if you have more details to add!

Obiwan2310, 2024-01-31:
Hello. I can confirm that this bug is still present in version 4.1.4, at least on Macintosh. The HTML code is generated incorrectly, regarding the class of the "span" tags, which are responsible for the colours in the code. The first part of the listing is usually coloured correctly, up to a certain line, then all the code only gets span tags with the class "COLOR_TODO" rather than the correct classes, such as "comment" etc.

Interestingly, the line where the bug hits varies between consecutive printouts of the same program, so it's obviously not related to the code to be printed. So far I couldn't find out what actually triggers it. Attached is a full example HTML file.

[Example.html.zip](https://github.com/thonny/thonny/files/14109562/Example.html.zip)

Hope this helps track down the beast :)

aivarannamaa, 2024-01-31:
Thanks! I'll investigate.

Obiwan2310, 2024-02-14:
I just noticed an interesting detail which hints at the origin of this bug, and which can also serve as a workaround: only the parts of a document that have been displayed on screen at least once are coloured correctly when printing via HTML. That means that if you open a long document, of which only the first page will be shown on screen, and you hit Print right away, only the first page will be in colour.

Therefore, as a workaround, scroll down to the bottom of the document after opening it, so that all the lines have been displayed on screen once, then hit Print afterwards. As a result, the entire document should print in colour.