thonny/thonny #1860: Cannot install thonny with apt on Ubuntu 21.04 (hirsute)

vonboth, 2021-06-10:
Hi!
I cannot install Thonny using apt in Ubuntu 21.04 (hirsute).

I get the message: Package "thonny" has no installation candidate

What am I missing?

aivarannamaa, 2021-06-10:
It looks like Thonny was not included in 21.04 apt repo. I suppose it's because it was excluded from Debian testing by mistake when 20.04 was prepared.

As an alternative, you can `pip install thonny` into a virtual environment or install Python+Thonny bundle with following command

`bash <(wget -O - https://thonny.org/installer-for-linux)`

vonboth, 2021-06-11:
Thank you for your quick answer and your help. I used Thonny in prev. Versions of Ubuntu and just didn't understand why I couldn't install it via apt on 21.04. I will use one of your suggested alternatives.
I will close this issue.
Thank you!

closed 2021-06-11 by vonboth