thonny/thonny #1861: kivy textinput produces blank screen

tamtienpl, 2021-06-11:
Dear Thonny Authors,
I have used Thonny IDE and it has been great.
Recently I use Kivy with TextInput and the result is a blank transparent screen.
(This happens on Thonny Linux Mint 20. It's OK when run Thonny on Windows)
When I run the file in computer Terminal Linux Mint 20, the result is good.
Code:
from kivy.config import Config
Config.set('kivy','keyboard_mode','systemanddock')
from kivy.app import App
from kivy.uix.textinput import TextInput # > BLANK SCREEN

class MyText(App):
    def build(self):
        t1=TextInput(size_hint=(.3,.1))
        return t1
MyText().run()

Please help. My gratefulness.
Nguyen Van Hoang

aivarannamaa, 2021-06-11:
Thank you for the report! 
I suspect the problem doesn't lie in Thonny, though, especially as the program works in Windows.

Have you asked about it in Kivy's forum?

aivarannamaa, 2021-07-06:
Please reopen if you have more details to add!

closed 2021-07-06 by aivarannamaa

DerPeterAusE, 2022-06-03:
Hello there,
i have the same Problem.
System ubuntu 20.04.4 LTS (64-bit)
Python 3.8.10 (64-bit)
Thonny 3.2.7
Kivy 2.1.0

main.py
`import kivy
kivy.require('2.1.0')
from datenbank import Datenbank

from kivy.uix.floatlayout import FloatLayout
from kivy.app import App
from kivy.properties import ObjectProperty, StringProperty
from kivy.uix.widget import Widget
from kivy.uix.textinput import TextInput

class Controller(FloatLayout):
    '''Create a controller that receives a custom widget from the kv lang file.

    Add an action to be called from the kv lang file.
    '''
    label_wid = ObjectProperty()
    info = StringProperty()

    def do_action(self):
        self.label_wid.text = 'My label after button press'
        self.info = 'New info text'
        
class ControllerApp(App):

    def build(self):
        return Controller(info='Hello world')

if __name__ == '__main__':
    ControllerApp().run()    
`
controller.kv
`#:kivy 1.0

<Controller>:
    label_wid: my_custom_label

    BoxLayout:
        orientation: 'vertical'
        padding: 20

        Button:
            id:the_button
            text: 'Hi Peter info is: ' + root.info
            on_press: root.do_action()

        Label:
            id: my_custom_label
            text: 'My label before button press'

        Label:
            id: my_custom_label_2
            text: 'Der Zweite'

        TextInput:
            id: testtext`

Without the last two lines in controller.kv its ok.
With them > blank screen.

A Test with the examples from kivy has no problem on the command line.
The same program in Thonny > blank screen.

Thanks a lot!

aivarannamaa, 2022-06-03:
@DerPeterAusE, thanks for the extra information!

I tried to reproduce with the [latest Thonny+Python bundle](https://github.com/thonny/thonny/releases/tag/v4.0.0b3) (Thonny 4.0.0b3 + Python 3.10) on Ubuntu 20.04, but got your example nicely working in Thonny (both via "Run current script" and "Run current script in terminal" commands).

I suggest to `apt remove thonny` and install the newer bundle instead via https://github.com/thonny/thonny/releases/download/v4.0.0b3/thonny-4.0.0b3.bash

The bundle still allows you to use your system Python for running user programs (click the selector at the lower-right corner of Thonny's main window).

DerPeterAusE, 2022-06-03:
@aivarannamaa , thanks for the quick Answer.

I removed thonny and removed kivy. Installed thonny with the script and add kivy with the plug-in manager.
If i start with the system Python, there is still a blank screen.
If i start with the Thonnys Python, is `ModuleNotFoundError: No module named 'kivy'` comming up.

here the kivy-log (maybe theres a hint

```
[INFO   ] Logger: Record log in /home/peter/.kivy/logs/kivy_22-06-03_15.txt
[INFO   ] Kivy: v2.1.0
[INFO   ] Kivy: Installed at "/home/peter/.local/lib/python3.8/site-packages/kivy/__init__.py"
[INFO   ] Python: v3.8.10 (default, Mar 15 2022, 12:22:08) 
[GCC 9.4.0]
[INFO   ] Python: Interpreter at "/usr/bin/python3.8"
[INFO   ] Logger: Purge log fired. Processing...
[INFO   ] Logger: Purge finished!
[INFO   ] Factory: 189 symbols loaded
[INFO   ] ImageLoaderFFPy: Using ffpyplayer 4.3.5
[INFO   ] Image: Providers: img_tex, img_dds, img_sdl2, img_pil, img_ffpyplayer 
[INFO   ] Window: Provider: sdl2
[INFO   ] GL: Using the "OpenGL" graphics system
[INFO   ] GL: Backend used <sdl2>
[INFO   ] GL: OpenGL version <b'4.6.0 NVIDIA 470.129.06'>
[INFO   ] GL: OpenGL vendor <b'NVIDIA Corporation'>
[INFO   ] GL: OpenGL renderer <b'NVIDIA GeForce GTX 1650/PCIe/SSE2'>
[INFO   ] GL: OpenGL parsed version: 4, 6
[INFO   ] GL: Shading version <b'4.60 NVIDIA'>
[INFO   ] GL: Texture max size <32768>
[INFO   ] GL: Texture max units <32>
[INFO   ] Window: auto add sdl2 input provider
[INFO   ] Window: virtual keyboard not allowed, single mode, not docked
[INFO   ] Text: Provider: sdl2
```

aivarannamaa, 2022-06-03:
The plug-in manager is not the right place for kivy. Use "Tools => Manage packages" instead.

DerPeterAusE, 2022-06-04:
remove plug-in, add packeges > Blank Screen :-(

next try to find more information: python3.10 main.py -m inspector
the output:
```
--- Logging error ---
 Traceback (most recent call last):
   File "/home/peter/.local/lib/python3.10/site-packages/kivy/input/providers/__init__.py", line 36, in <module>
     import kivy.input.providers.probesysfs
   File "/home/peter/.local/lib/python3.10/site-packages/kivy/input/providers/probesysfs.py", line 50, in <module>
     import ctypes
   File "/home/peter/apps/thonny/lib/python3.10/ctypes/__init__.py", line 8, in <module>
     from _ctypes import Union, Structure, Array
 ImportError: libffi.so.6: cannot open shared object file: No such file or directory
 
 During handling of the above exception, another exception occurred:
 
 Traceback (most recent call last):
   File "/home/peter/apps/thonny/lib/python3.10/logging/__init__.py", line 1100, in emit
     msg = self.format(record)
   File "/home/peter/apps/thonny/lib/python3.10/logging/__init__.py", line 943, in format
     return fmt.format(record)
   File "/home/peter/.local/lib/python3.10/site-packages/kivy/logger.py", line 291, in format
     record = copy.deepcopy(record)
   File "/home/peter/apps/thonny/lib/python3.10/copy.py", line 172, in deepcopy
     y = _reconstruct(x, memo, *rv)
   File "/home/peter/apps/thonny/lib/python3.10/copy.py", line 271, in _reconstruct
     state = deepcopy(state, memo)
   File "/home/peter/apps/thonny/lib/python3.10/copy.py", line 146, in deepcopy
     y = copier(x, memo)
   File "/home/peter/apps/thonny/lib/python3.10/copy.py", line 231, in _deepcopy_dict
     y[deepcopy(key, memo)] = deepcopy(value, memo)
   File "/home/peter/apps/thonny/lib/python3.10/copy.py", line 146, in deepcopy
     y = copier(x, memo)
   File "/home/peter/apps/thonny/lib/python3.10/copy.py", line 211, in _deepcopy_tuple
     y = [deepcopy(a, memo) for a in x]
   File "/home/peter/apps/thonny/lib/python3.10/copy.py", line 211, in <listcomp>
     y = [deepcopy(a, memo) for a in x]
   File "/home/peter/apps/thonny/lib/python3.10/copy.py", line 161, in deepcopy
     rv = reductor(4)
 TypeError: cannot pickle 'traceback' object
 Call stack:
   File "/home/peter/Phyton_Thonny/erste_virtuelle/main.py", line 27, in <module>
     ControllerApp().run()
   File "/home/peter/.local/lib/python3.10/site-packages/kivy/app.py", line 955, in run
     runTouchApp()
   File "/home/peter/.local/lib/python3.10/site-packages/kivy/base.py", line 567, in runTouchApp
     _runTouchApp_prepare(widget=widget)
   File "/home/peter/.local/lib/python3.10/site-packages/kivy/base.py", line 484, in _runTouchApp_prepare
     from kivy.input import MotionEventFactory, kivy_postproc_modules
   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
   File "/home/peter/.local/lib/python3.10/site-packages/kivy/input/__init__.py", line 30, in <module>
     from kivy.input.postproc import kivy_postproc_modules
   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
   File "/home/peter/.local/lib/python3.10/site-packages/kivy/input/postproc/__init__.py", line 15, in <module>
     from kivy.input.postproc.calibration import InputPostprocCalibration
   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
   File "/home/peter/.local/lib/python3.10/site-packages/kivy/input/postproc/calibration.py", line 51, in <module>
[... 70 lines cut ...]
   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
   File "/home/peter/.local/lib/python3.10/site-packages/kivy/input/postproc/__init__.py", line 15, in <module>
     from kivy.input.postproc.calibration import InputPostprocCalibration
   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
   File "/home/peter/.local/lib/python3.10/site-packages/kivy/input/postproc/calibration.py", line 51, in <module>
     from kivy.input import providers
   File "<frozen importlib._bootstrap>", line 1078, in _handle_fromlist
   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
[... cut ...] ", line 883, in exec_module
   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
   File "/home/peter/.local/lib/python3.10/site-packages/kivy/input/providers/__init__.py", line 44, in <module>
     Logger.exception(err)
 Message: 'Input: MTDev is not supported by your version of linux'
 Arguments: ()
[WARNING] [Base        ] Unknown <probesysfs> provider
[INFO   ] [Base        ] Start application main loop
[INFO   ] [GL          ] NPOT texture support is available
```

DerPeterAusE, 2022-06-04:
if i run the inspector, the Screen is visible.

DerPeterAusE, 2022-06-04:
make a new virtuellMashiene with xubuntu, install thonny per script and  add kivy with "Manage package"
In Thonny the program shows the Screen.
Maybe the kivylog helps:
```
[INFO   ] Logger: Record log in /home/peter/.kivy/logs/kivy_22-06-04_1.txt
[INFO   ] Kivy: v2.1.0
[INFO   ] Kivy: Installed at "/home/peter/.local/lib/python3.10/site-packages/kivy/__init__.py"
[INFO   ] Python: v3.10.4 (main, Apr 10 2022, 09:00:53) [GCC 5.4.0 20160609]
[INFO   ] Python: Interpreter at "/home/peter/apps/thonny/bin/python3.10"
[INFO   ] Logger: Purge log fired. Processing...
[INFO   ] Logger: Purge finished!
[INFO   ] Factory: 189 symbols loaded
[INFO   ] Image: Providers: img_tex, img_dds, img_sdl2 (img_pil, img_ffpyplayer ignored)
[INFO   ] Window: Provider: sdl2
[INFO   ] GL: Using the "OpenGL" graphics system
[INFO   ] GL: Backend used <sdl2>
[INFO   ] GL: OpenGL version <b'4.5 (Compatibility Profile) Mesa 22.0.1'>
[INFO   ] GL: OpenGL vendor <b'Mesa/X.org'>
[INFO   ] GL: OpenGL renderer <b'llvmpipe (LLVM 13.0.1, 256 bits)'>
[INFO   ] GL: OpenGL parsed version: 4, 5
[INFO   ] GL: Shading version <b'4.50'>
[INFO   ] GL: Texture max size <16384>
[INFO   ] GL: Texture max units <32>
[INFO   ] Window: auto add sdl2 input provider
[INFO   ] Window: virtual keyboard not allowed, single mode, not docked
[INFO   ] Text: Provider: sdl2
[INFO   ] Clipboard: Provider: sdl2(['clipboard_xclip', 'clipboard_xsel', 'clipboard_dbusklipper', 'clipboard_gtk3'] ignored)
[CRITICAL] Cutbuffer: Unable to find any valuable Cutbuffer provider. Please enable debug logging (e.g. add -d if running from the command line, or change the log level in the config) and re-run your app to identify potential causes
xclip - FileNotFoundError: [Errno 2] No such file or directory: 'xclip'
  File "/home/peter/.local/lib/python3.10/site-packages/kivy/core/__init__.py", line 59, in core_select_lib
    mod = importlib.__import__(name='{2}.{0}.{1}'.format(
  File "<frozen importlib._bootstrap>", line 1129, in __import__
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/peter/.local/lib/python3.10/site-packages/kivy/core/clipboard/clipboard_xclip.py", line 17, in <module>
    p = subprocess.Popen(['xclip', '-version'], stdout=subprocess.PIPE,
  File "/home/peter/apps/thonny/lib/python3.10/subprocess.py", line 966, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/home/peter/apps/thonny/lib/python3.10/subprocess.py", line 1842, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)

xsel - FileNotFoundError: [Errno 2] No such file or directory: 'xsel'
  File "/home/peter/.local/lib/python3.10/site-packages/kivy/core/__init__.py", line 59, in core_select_lib
    mod = importlib.__import__(name='{2}.{0}.{1}'.format(
  File "<frozen importlib._bootstrap>", line 1129, in __import__
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/peter/.local/lib/python3.10/site-packages/kivy/core/clipboard/clipboard_xsel.py", line 16, in <module>
    p = subprocess.Popen(['xsel'], stdout=subprocess.PIPE)
  File "/home/peter/apps/thonny/lib/python3.10/subprocess.py", line 966, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/home/peter/apps/thonny/lib/python3.10/subprocess.py", line 1842, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)

[INFO   ] Base: Start application main loop
[INFO   ] GL: NPOT texture support is available
```

aivarannamaa, 2022-07-06:
Sorry for the late reply! From the log it looks like Kivy requires `xsel` and/or `xclip` programs to be present (probably for clipboard management). Try executing `sudo apt install xsel xclip`.

DerPeterAusE, 2022-07-07:
This fix the problem.
Thanks a lot!