thonny/thonny #1862: Error launching Thonny: _tkinter.TclError: couldn't recognize data in image file "...thonny.png"

kiwioverseas, 2021-06-12:
I installed Thonny using pip in Oracle Linux because it's not available in yum.

When I try to launch it using `./thonny` I get this error:

```INFO    thonny: Thonny version: 3.3.10
Traceback (most recent call last):
  File "./thonny", line 11, in <module>
    sys.exit(launch())
  File "/usr/local/lib/python3.6/site-packages/thonny/__init__.py", line 178, in launch
    if not _check_welcome():
  File "/usr/local/lib/python3.6/site-packages/thonny/__init__.py", line 151, in _check_welcome
    win = FirstRunWindow(mgr)
  File "/usr/local/lib/python3.6/site-packages/thonny/first_run.py", line 30, in __init__
    self.logo = tk.PhotoImage(file=logo_file)
  File "/usr/lib64/python3.6/tkinter/__init__.py", line 3545, in __init__
    Image.__init__(self, 'photo', name, cnf, master, **kw)
  File "/usr/lib64/python3.6/tkinter/__init__.py", line 3501, in __init__
    self.tk.call(('image', 'create', imgtype, name,) + options)
_tkinter.TclError: couldn't recognize data in image file "/usr/local/lib/python3.6/site-packages/thonny/res/thonny.png"

aivarannamaa, 2021-06-12:
Thank you for the report!

I suspect you don't have recent enought Tcl/Tk. Thonny requires Tk 8.6.

You could try Thonny's binary bundle, which includes Python 3.7 and Tk 8.6:

```
bash <(wget -O - https://thonny.org/installer-for-linux)
```

kiwioverseas, 2021-06-12:
> Thank you for the report!
> 
> I suspect you don't have recent enought Tcl/Tk. Thonny requires Tk 8.6.
> 
> You could try Thonny's binary bundle, which includes Python 3.7 and Tk 8.6:

Thanks. I'm using Oracle Linux on ARM and can't find an ARM prebuilt. I have Python 3.6 & Tkinter 8.5.13. Is this combination not supported?

aivarannamaa, 2021-06-12:
Nope, Tk 8.5 is not supported. 

The latest Thonny version that supports it, is 2.1.22, so you could try `pip3 install thonny==2.1.22`

kiwioverseas, 2021-06-12:
> The latest Thonny version that supports it, is 2.1.22, so you could try `pip3 install thonny==2.1.22`

Thank you! This version worked.

closed 2021-06-12 by kiwioverseas