thonny/thonny #1863: HeadsUP Pico time syncing

kevindawson, 2021-06-13:
- Thonny 3.3.10
- RP Pico
- micropython rp2-pico-20210613-unstable-v1.15-211-g3ab8806c0
  - includes #https://github.com/micropython/micropython/pull/6928

Thonny Pico time syncing now fails
```python
import utime
print(utime.localtime())
```
giving
`(2021,1,1...)`
Previous uf2 v1.15-207 worked ok

aivarannamaa, 2021-06-13:
Thanks for the notification!

I'll try to make time sync work properly over different versions and ports of MicroPython, but it's not an easy job given all this confusion regarding the formats.

kevindawson, 2021-06-13:
@aivarannamaa keep up the good work

aivarannamaa, 2021-06-18:
It looks like time syncing works again with just-released MicroPython 1.16

closed 2021-06-18 by aivarannamaa

kevindawson, 2021-06-18:
@aivarannamaa 
as Thonny is still 3.3.10
what will happen on a Sunday

```python
rtc = machine.RTC()
rtc.datetime((2021, 6, 20, 7, 11, 11, 11, 0))
print("rtc:",rtc.datetime())
```
will then generate:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 22] EINVAL
```
thoughts?

aivarannamaa, 2021-06-19:
Right. There was discussion about ignoring provided weekday and computing the correct one, and when I tested 1.16 yesterday, I totally forgot to test it with Sunday.

I'll fix it under #1868 and hopefully can find time to make a release tonight.

kevindawson, 2021-06-19:
weekday 0-6 (with Monday being zero)