thonny/thonny #1864: Backend terminated or disconnected. Use 'Stop/Restart' to restart when having a memory error

Cr0c0s, 2021-06-14:
Hello,
I am using Thonny 3.3.6 with Python 3.7.9. My current programme tries to concatenat lots of .csv files when this error message pops : 

> Traceback (most recent call last):
  File "C:\Users\vincent.nourry\AppData\Local\Programs\Thonny\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "C:\Users\vincent.nourry\AppData\Local\Programs\Thonny\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\vincent.nourry\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\__main__.py", line 43, in <module>
    MainCPythonBackend(target_cwd).mainloop()
  File "C:\Users\vincent.nourry\AppData\Local\Programs\Thonny\lib\site-packages\thonny\backend.py", line 72, in mainloop
    self._handle_normal_command(msg)
  File "C:\Users\vincent.nourry\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 254, in _handle_normal_command
    self.send_message(real_response)
  File "C:\Users\vincent.nourry\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 920, in send_message
    self._original_stdout.write(serialize_message(msg) + "\n")
  File "C:\Users\vincent.nourry\AppData\Local\Programs\Thonny\lib\site-packages\thonny\common.py", line 220, in serialize_message
    return MESSAGE_MARKER + ascii(msg)
  File "C:\Users\vincent.nourry\AppData\Local\Programs\Thonny\lib\site-packages\thonny\common.py", line 88, in __repr__
    return "{}({})".format(self.__class__.__name__, ", ".join(items))
  File "C:\Users\vincent.nourry\AppData\Local\Programs\Thonny\lib\site-packages\thonny\common.py", line 87, in <genexpr>
    items = ("{}={}".format(k, repr(self.__dict__[k])) for k in keys)
MemoryError
Backend terminated or disconnected. Use 'Stop/Restart' to restart.

Thonny has works fine for now on my computer, can you help me ?

aivarannamaa, 2021-06-14:
Thank you for the report!

Please check, whether you can run your program with "Run => Run current script in Terminal" ?

Cr0c0s, 2021-06-14:
When I launch the programme with Run current script in Terminal, the same error occure

aivarannamaa, 2021-06-14:
Hmm... in this case I suspect that you simply run out of memory and it's not Thonny's fault.

Can you show me your code? Also, please give me an indication about the size of your input data (number of files, rows, columns). Maybe I can spot an inefficiency in your code.

Cr0c0s, 2021-06-14:
I try to open a 500 columns by 300 00 lignes file, so it is probably this the problem

aivarannamaa, 2021-06-14:
This data size is not problem by itself, but it can be problem if you are trying to convert into regular Python list or list of dictionaries or something like this. Pandas/Numpy data structures require much less memory than regular Python types.

Cr0c0s, 2021-06-14:
Anyway, I have solved my problem by splitting my file, thanks for your help.
And thanks for your work on thonny, it's really helpful :)
Chears

aivarannamaa, 2021-06-14:
Thanks and good luck!

closed 2021-06-14 by aivarannamaa