thonny/thonny #1865: Thonny shell fail to run prompt_toolkit program

PyHubs, 2021-06-14:
It seems the thonny ide fails too run code with prompt_toolkit moudle. Thus shell doesnt really work.

(I just was using the prompt_toolkit session = PromptSession() function, it works when i run it with cmd.exe or a terminal emulator)

Error code
```python
Python 3.9.5 (C:/Users/izooi_2ppgdho/AppData/Local/Programs/Python/Python39/python.exe)
>>> %Run bootup.py
Traceback (most recent call last):
  File "C:\Workspace-CLI\bootup.py", line 9, in <module>
    session = PromptSession()
  File "C:\Users\izooi_2ppgdho\AppData\Local\Programs\Python\Python39\lib\site-packages\prompt_toolkit\shortcuts\prompt.py", line 468, in __init__
    self.app = self._create_application(editing_mode, erase_when_done)
  File "C:\Users\izooi_2ppgdho\AppData\Local\Programs\Python\Python39\lib\site-packages\prompt_toolkit\shortcuts\prompt.py", line 719, in _create_application
    application: Application[_T] = Application(
  File "C:\Users\izooi_2ppgdho\AppData\Local\Programs\Python\Python39\lib\site-packages\prompt_toolkit\application\application.py", line 277, in __init__
    self.output = output or session.output
  File "C:\Users\izooi_2ppgdho\AppData\Local\Programs\Python\Python39\lib\site-packages\prompt_toolkit\application\current.py", line 70, in output
    self._output = create_output()
  File "C:\Users\izooi_2ppgdho\AppData\Local\Programs\Python\Python39\lib\site-packages\prompt_toolkit\output\defaults.py", line 73, in create_output
    return Win32Output(stdout, default_color_depth=color_depth_from_env)
  File "C:\Users\izooi_2ppgdho\AppData\Local\Programs\Python\Python39\lib\site-packages\prompt_toolkit\output\win32.py", line 107, in __init__
    info = self.get_win32_screen_buffer_info()
  File "C:\Users\izooi_2ppgdho\AppData\Local\Programs\Python\Python39\lib\site-packages\prompt_toolkit\output\win32.py", line 211, in get_win32_screen_buffer_info
    raise NoConsoleScreenBufferError
prompt_toolkit.output.win32.NoConsoleScreenBufferError: No Windows console found. Are you running cmd.exe?
>>> %Run bootup.py
Traceback (most recent call last):
  File "C:\Workspace-CLI\bootup.py", line 11, in <module>
    print(HTML('<style fg="lightgreen">Workspace CLI - Bootup manager</style>'))
  File "C:\Users\izooi_2ppgdho\AppData\Local\Programs\Python\Python39\lib\site-packages\prompt_toolkit\shortcuts\utils.py", line 103, in print_formatted_text
    output = get_app_session().output
  File "C:\Users\izooi_2ppgdho\AppData\Local\Programs\Python\Python39\lib\site-packages\prompt_toolkit\application\current.py", line 70, in output
    self._output = create_output()
  File "C:\Users\izooi_2ppgdho\AppData\Local\Programs\Python\Python39\lib\site-packages\prompt_toolkit\output\defaults.py", line 73, in create_output
    return Win32Output(stdout, default_color_depth=color_depth_from_env)
  File "C:\Users\izooi_2ppgdho\AppData\Local\Programs\Python\Python39\lib\site-packages\prompt_toolkit\output\win32.py", line 107, in __init__
    info = self.get_win32_screen_buffer_info()
  File "C:\Users\izooi_2ppgdho\AppData\Local\Programs\Python\Python39\lib\site-packages\prompt_toolkit\output\win32.py", line 211, in get_win32_screen_buffer_info
    raise NoConsoleScreenBufferError
prompt_toolkit.output.win32.NoConsoleScreenBufferError: No Windows console found. Are you running cmd.exe?
>>> 
```

aivarannamaa, 2021-06-14:
Thank you for the report! 

Unfortunately this is where Thonny's scope ends -- Thonny's shell is not a terminal emulator. As a work-around you may want to use "Run => Select current script in terminal".

PyHubs, 2021-06-14:
Alright @aivarannamaa thanks for the reply :)

closed 2021-06-14 by PyHubs