thonny/thonny #1876: Add .mpy support

The-Bow-Hunter, 2021-06-24:
Add the support to encode to mpy and easily send them to the board.

So far it is necessary to create those by yourself, then locate it on the left and then send it to the board. It would be easier if it could be done simular to open and then just be send over, or to even have the possibility to save python files as mpy using save as.

aivarannamaa, 2021-06-25:
Thank you for the suggestion!

I've considered this and I'll keep it in my TODO list.

aivarannamaa, 2021-06-30:
When implementing this, https://pypi.org/project/mpy-cross/ will be probably useful.

Note, that the task is not trivial, though, as different MicroPython version may require different versions of mpy-cross and different boards may require different parameters for inline assembler etc.

veni-vidi-code, 2021-07-01:
https://github.com/micropython/micropython/tree/master/mpy-cross
Could be useful aswell if you prefer to build it yourself to be able to tweek it