thonny/thonny #1882: Installing mail module erroe

narunkumark, 2021-06-29:
Getting bellow error while installing mail module.

--compile --install-headers 'C:\Users\a_nataraj\AppData\Roaming\Python\Python37\Include\main' Check the logs for full command output.
Process returned with code 1

aivarannamaa, 2021-06-29:
Is it this package: https://pypi.org/project/Mail/ ?

Considering the last release is from 2010, I don't find it likely that it works in Python 3

Python 3 standard library already has `email` module. Are you sure you need to install something extra?

narunkumark, 2021-06-29:
sorry that was typo.. its main. corrected the mistake.

aivarannamaa, 2021-06-29:
Do you really mean https://pypi.org/project/main/ ?

I suspect you actually meant something else than "installing main module", but I can't see what it is.

Can you tell me in more details what you were trying to achieve, what steps did you take and what happened?

closed 2021-07-06 by aivarannamaa