thonny/thonny #1883: Large Int Object IDs not accurately represented

Andreilg, 2021-06-30:
It is my understanding that in python, small Int objects ([-5:256]) are not garbage collected and new instances of Int objects of the same value in that range are never created, thus there is only one "5" object for example. However for numbers outside that range, python will create a new Int object every time an Int object is explicitly bound. This means that they should have different IDs.

However, the heap in Thonny is showing that the value IDs are identical, while checking the IDs directly in the Python console shows they are different. 
[image]
[image]

aivarannamaa, 2021-06-30:
Did you try running the *script* of two `999999999` objects? According to my experiment, the id will be same.

I suppose Python's compiler optimises memory use when it sees two identical int literals in the source code and uses same runtime object for both. When you create two objects one by one as separate REPL inputs, Python also compiles the statements separately.

Andreilg, 2021-07-02:
I just checked a script and you are correct. In retrospect it makes sense that a script would be more optimized in this respect.

Thank you for your time.

closed 2021-07-02 by Andreilg