thonny/thonny #1887: Modal windows

JohnRains, 2021-07-05:
I've just installed Thonny on my MacBook Pro, MacOS Big Sur v11.4, to do development with the RP-Pico. 
I run with 2 external 23" monitors and if I launch Thonny on one of these external monitors then do anything that loads a modal window (such as Help) the modal flashes on the screen then disappears. Because it's modal, it stops any other function until closed. If I drag the main Thonny window toward the primary screen, the modal becomes visible on the primary screen and then can be closed. I have verified this with Thonny 3.3.11 and 3.3.10-alt.

aivarannamaa, 2021-07-06:
Thank you for the report!

Do I understand correctly, that Thonny works properly, when launched from the main (laptop?) screen, but misbehaves when launched from the external screen? Does display arrangement affect the problem / does the same problem occurs on both of your external screens? 

Unfortunately I could not reproduce this with Catalina and single external display, but I noticed that Thonny had problems restoring window position after being closed on the external display.

Do you happen to have official Python installation on your computer? Then you could try, whether IDLE (which uses same UI toolkit as Thonny) misbehaves the same way.

JohnRains, 2021-07-06:
I believe you have hit on the issue.

My screen arrangement is with the laptop in the center with one monitor to the left and to other to the right. 

I believe screen position values are relative to the primary so that means the left monitor uses negative x offsets. To verify this, I placed the Thonny main window in the center of the left monitor. I then clicked the file open icon and no modal appeared as shown in screenshot 1. I then started to drag the main window to the left and the file open mode dialog was pulled onto the left monitor as shown in screenshot 2. I then repeated by placing the Thonny main window on the hard right of the left monitor and repeated the file open. Dragging the main window left shows the file open dialog on the immediate right side of the main window as shown in screenshot 3. In other words, the modal is receiving an initial x position of zero (presumably an unsigned int). 

I can live with this.

John.

aivarannamaa, 2022-05-17:
Assuming it got fixed in https://github.com/thonny/thonny/releases/tag/v4.0.0b3, as confirmed by the reporter of a similar issue (#1947)

closed 2022-05-17 by aivarannamaa