thonny/thonny #1889: ESP8266 lost communication with Thonny after firmware installation

RXu2013, 2021-07-09:
Firmware (esp8266-1m-20210618-v1.16.bin) was installed successfully on to the ESP8266 according to the message in Thonny IDE. After that the ESP8266 becomes busy and can not communication with REPL. 
The following message is displayed in Shell:

Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - use Stop/Restart to interrupt more and enter REPL.

All this happened after recent Windows 10 updates.
USB-SERIAL CH340 was used for installing the firmware.

simojo, 2021-07-10:
It might be worth first making sure there isn't any code being executed on startup and that it's not relapsing into some kind of boot mode.

aivarannamaa, 2021-07-10:
@RXu2013, did you erase flash while installing the firmware? If yes, then we don't need too suspect the startup code causing problems.

Are you sure the firmware you installed is suitable for your board? Are you sure you used proper flash mode? Even if Thonny (actually `esptool`) said the firmware installation succeeded, we can't be sure the firmware and parameters were suitable.

Also, make sure you were using ESP8266 back-end in Thonny, not ESP32-s. The firmware dialog looks identical (apart from the title), but the firmware gets installed to different location in either case.

RXu2013, 2021-07-10:
Hi Aivar,

Thanks for your reply.
“Erase flash before installing” box was ticked when installing the firmware.
***@***.***D775C5.35B2DB00]

I installed the firmware on the same ESP-01S a month ago with no problems on communication between ESP-01s and REPL.
After installed the latest windows 10 updates a couple of days ago, I tried to reinstall the firmware and after installation the comms lost between ESP8266 and REPL. I am not sure the problem is caused by the windows updates or something else.

Regards

Rob

aivarannamaa, 2021-07-10:
You should try connecting with Putty as well (try Ctrl+C or Ctrl+B after connecting). If you still don't get REPL, then try https://forum.micropython.org/viewforum.php?f=16

closed 2021-08-01 by aivarannamaa