thonny/thonny #1892: Import serial from pyserial issue

Nutakki2259, 2021-07-11:
Hi all,

I am just running this 'Import serial' and obtaining this error `NameError: name 'IOError' isn't defined` when I connect Raspberry Pi Pico to the PC. I am unable to solve this issue. As the same format works fine in Spyder.

All suggestions are accepted.

aivarannamaa, 2021-07-12:
Thank you for the report!

Do you try to run `import serial` on your PC (for creating custom communication with a serial device) or on Pico (I don't know a library with module `serial` for MicroPython)?

Do you get this error on Pico or PC (ie. Python 3) ?

Nutakki2259, 2021-07-12:
I got this error in Pico. I am just importing the serial library and running which shows this error in Pico but not in PC.

aivarannamaa, 2021-07-12:
Could you give me the whole script you're trying to run on your Pico?

Nutakki2259, 2021-07-12:
This is the whole code on Pico:

`#!/usr/bin/env python3
import serial
if __name__ == '__main__':
    ser = serial.Serial('COM8', 9600, timeout=1)
    ser.flush()
    while True:
        if ser.in_waiting > 0:
            line = ser.readline().decode('utf-8').rstrip()
            print(line)`

aivarannamaa, 2021-07-13:
These pieces of code make sense in the Windows PC side, not on Pico, because `serial` is a library for CPython and COM-ports are Windows thing.

Nutakki2259, 2021-07-13:
Yeah, I know but the `serial` library is working on Anaconda Spyder but not on Thonny with the same Pico.

aivarannamaa, 2021-07-13:
I suspect I just misunderstood you and you actually want to run your script on CPython and have it talk to your Pico.

In this case, do you have Python 3 back-end selected in Thonny? Can you make a screenshot of your whole Thonny window, which shows the code, error message and the back-end descriptor in the lower-right corner of the window?

Nutakki2259, 2021-07-13:
Here are the two program structure screenshots that show errors.

[image]

[image]

aivarannamaa, 2021-07-13:
I see that you are trying to connect to a COM port on Pico, but that's not possible -- COM ports are a Windows thing, not Pico's MicroPython's. Also, it looks that you have installed PySerial on your Pico, but it won't work there -- it is meant to be run on a regular computer.

You probably want to switch Thonny's back-end to CPython (eg. "The same interpreter which runs Thonny") and then run your code from your local disk.

Nutakki2259, 2021-07-20:
Thank you for your reply. Can I have an example to run the `pyserial` accessing the port that makes the Pico run the program?

aivarannamaa, 2021-07-20:
I suggest to ask this at https://www.raspberrypi.org/forums/viewforum.php?f=146 or https://forum.micropython.org/viewforum.php?f=21

closed 2021-08-01 by aivarannamaa