thonny/thonny #1893: startup screen

sk8nfool, 2021-07-11:
I just installed Thonny (V3.3.?) and when I started it is popped up a window asking what language and initial settings I want. What is this??? I can't seem to find any mention of this anywhere! I'm running this on a laptop running Ubuntu 18.04. are the initial settings for the OS I'm running Thonny on or the platform I'm building an application for? The options (standard/Raspberry PI) are ambigous. I'm trying to develop an app for the RPi Pico. Here's a screenshot of this window (I hope)
[image]

aivarannamaa, 2021-07-12:
Thank you for the report!

You can develop for Pico with both settings. With Raspberry settings Thonny starts in simple mode, different theme and a couple of other tweaks Raspberry people chose for default settings on Raspberry Pi. I recommend choosing "Standard".

I'll try to make it more clear in the next version.

sk8nfool, 2021-08-13:
Ah - this sets Thonny's host defaults not the target defaults. That's what needs to be a little more explicit. This is especially true for documentation.

aivarannamaa, 2021-08-13:
Let's keep this open so I remember to make this dialog more clear.

aivarannamaa, 2024-05-30:
The options are now:

* Regular
* Simplified
* Simplified, with Raspberry Pi theme and line-based debugger

closed 2024-05-30 by aivarannamaa