thonny/thonny #1895: Localization files fail to compile

befeleme, 2021-07-12:
For the purposes of packaging Thonny in Fedora I wanted to compile translation files during the build time (rather than provide precompiled binary files).
For that, I used the same command there is in `thonny/locale/compile_mo.bat`, but the compilation failed with errors:

```
+ pybabel compile -d thonny/locale/ -D thonny
<snip>
error: thonny/locale/de_DE/LC_MESSAGES/thonny.po:982: placeholders are incompatible
error: thonny/locale/de_DE/LC_MESSAGES/thonny.po:1412: positional format placeholders are unbalanced
error: thonny/locale/hu_HU/LC_MESSAGES/thonny.po:1388: incompatible format for placeholder 1: 's' and 'i' are not compatible
error: thonny/locale/pt_BR/LC_MESSAGES/thonny.po:977: placeholders are incompatible
4 errors encountered.
```

aivarannamaa, 2021-07-12:
Thank you for the report!

These translations originate from https://poeditor.com/join/project/Gh188fdYH6. I'll check later, what's wrong there. 

Do you notice something suspicious on these lines in the po files?

befeleme, 2021-07-13:
The translations can be corrected even by the people not knowing the destination language. 
It's either formatting (missing/typo in %s placeholder) or inconsistent lines count in source and translation. 
I'd correct it right away in the files, but seeing there is a dedicated service to that, I didn't have time to explore it yet.

befeleme, 2021-07-16:
Hey, I've edited three of the problematic strings in the respective translations.
This one needs translation of the last line but I can't provide it: 
`error: thonny/locale/pt_BR/LC_MESSAGES/thonny.po:977: placeholders are incompatible`

aivarannamaa, 2021-07-19:
Thank you! I pulled these fixes in (together with Google-translated last line of the Portuguese one). They will be part of Thonny 3.3.12.

I'll add `pybabel compile` as part of my release in order to catch similar errors earlier.

BTW. I used to compile mo files myself, but Babel had problems with Chinese ones. I suggest you check whether your mo-s for Chinese look right.

closed 2021-07-19 by aivarannamaa