thonny/thonny #1903: Go to line number.

jrs69, 2021-07-19:
I'm new to Thonny, all good!, except I would love a shortcut to go to a specific line, e.g. ctrl+g or ctrl-l etc.

I'm a retired programmer just doing hobby stuff, but were I still working it would be a problem.

aivarannamaa, 2021-07-19:
Thank you for the suggestion!

This would be a nice and non-obtrusive addition. I'll add it into my TODO list.

Related to #1665

neil9, 2021-08-07:
Hear hear. It would be VERY nice to be able to go directly to the line number rather than having to scroll through the whole code.

bigrom26, 2022-01-06:
I try go to selected line but not work Ctrl-g and not in menu. 
How correct this problem ? 
I work on Win10.

aivarannamaa, 2022-01-08:
@bigrom26, this feature isn't implemented yet.

aivarannamaa, 2022-01-31:
Implemented in #2001

closed 2022-01-31 by aivarannamaa