thonny/thonny #1908: ttk system theme option

l33tlinuxh4x0r, 2021-07-21:
Could we please get a theme included with Thonny that uses the system default ttk theme? Rather than creating custom themes why not just use the default ttk themes that are installed on the system. This way the user could select (system wide) what they want tk to look like and Thonny would follow suit.

Label: proposals for changes and new features

aivarannamaa, 2021-07-21:
Thank you for the suggestion?

What do you mean by system default ttk theme? How does one set it? I assume it's a Linux thing and separate from general system theme.

l33tlinuxh4x0r, 2021-07-21:
Yeah a bit naive of me. On Linux anyway you can use a program called ttk theme chooser. https://github.com/Geballin/TTK-Theme-chooser You can also install custom themes. But I usually just use the GTK theme... gtkttk https://github.com/Geballin/gtkTtk which makes tk applications appear as native gtk applications. I believe that there is a theme for qt as well.

For what it is worth, I tried editing the Thonny source code to make a blank theme to see if it would grab from the default but it just threw errors.

l33tlinuxh4x0r, 2021-07-21:
It looks like Thonny already uses ttk. ttk is supposed to grab the system theme and make tk look like the system theme by default. Here is an example app that shows the difference. 

```python
import tkinter as tk
from tkinter import ttk

root = tk.Tk()

# classic tk label
tk.Label(root, text='Classic Label').pack()
# change the tk to ttk
ttk.Label(root, text='Themed Label').pack()

# classic tk button
button = tk.Button(root, text="Classic Button").pack()
# change the tk to ttk
button2 = ttk.Button(root, text="Themed Button").pack()

root.mainloop()```

I'll upload a screen shot as well.
![Screenshot from 2021-07-21 16-31-55](https://user-images.githubusercontent.com/1632838/126556078-95f64cd4-18bd-4c3f-b53d-c6296a877b62.png)

aivarannamaa, 2021-07-22:
I wasn't aware of the gtkTtk theme -- thank you for bringing this to my attention! It looks promising, but according to my quick test (via https://github.com/TkinterEP/python-gttk) it has some issues and can't be used in Thonny as-is. I hope I can find time to tweak it later. Even then I probably can't make it work in pip-installed Thonny, but I might be able to include it in binary Linux builds of Thonny.

Regarding setting the default Ttk theme -- I also wasn't aware of this possibility. Still, I don't think I want to make Thonny choose an unknown theme automatically. All Ttk themes Thonny currently offers, come with extra tweaks implemented in Thonny's code. When I find time, I may extend the selection, but I want to test and possibly tweak each new theme first.