thonny/thonny #1918: thonny.backend: Exception in _trace

Caraot, 2021-07-28:
Hello, Team. I hope you all are Good

Recently i am using the debbugin and when I am interenting insiside of every statement i get this error.
I reinstalled the application and its not working. The version 3.1 does not present this error.
I am absolute beginner, so I dont have any Idea what this means. 
I thanks all in advance for the help

ERROR   thonny.backend: Exception in _trace
Traceback (most recent call last):
  File "C:\Users\h\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1787, in _trace
    return self._trace_and_catch(frame, event, arg)
  File "C:\Users\h\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1832, in _trace_and_catch
    self._handle_progress_event(frame.f_back, event, marker_function_args, node)
  File "C:\Users\h\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1891, in _handle_progress_event
    self._respond_to_commands()
  File "C:\Users\h\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 2018, in _respond_to_commands
    self._fetch_next_debugger_command(frame)
  File "C:\Users\h\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 1444, in _fetch_next_debugger_command
    cmd = self._backend._fetch_next_incoming_message()
  File "C:\Users\h\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 173, in _fetch_next_incoming_message
    self._read_one_incoming_message()
  File "C:\Users\h\AppData\Local\Programs\Thonny\lib\site-packages\thonny\backend.py", line 119, in _read_one_incoming_message
    line = self._read_incoming_msg_line()
  File "C:\Users\h\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 203, in _read_incoming_msg_line
    return self._original_stdin.readline()
  File "C:\Users\humbe\AppData\Local\Programs\Thonny\lib\codecs.py", line 319, in decode
    def decode(self, input, final=False):
  File "C:\Users\h\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 163, in signal_handler
    raise KeyboardInterrupt("Execution interrupted")
KeyboardInterrupt: Execution interrupted

aivarannamaa, 2021-07-29:
Thank you for the report!

From the stacktrace it looks like you have pressed Ctrl+C during debugging. Did you? (I'm not saying you shouldn't)

Does it happens with all programs or with a specific one?

Caraot, 2021-07-30:
Hello, 
I just do it clicking on the debugging Icon. 
And all its happen in all the programs

aivarannamaa, 2021-07-31:
Could you give me one simple example program and the list of button clicks or keypresses you perform to reproduce this problem?

Caraot, 2021-08-02:
one of the example was:
lista = [1,2,3,4,5,6]
when I try to debugging step by step to see how the list is form it  showed that error message

aivarannamaa, 2021-08-03:
I could not reproduce this problem in Linux. Trying Windows later.

aivarannamaa, 2021-08-07:
I also could not reproduce it in Windows 10, using Thonny 3.3.13

Could you list exact actions (mouse clicks and keypresses) for reproducing it?

closed 2021-11-06 by aivarannamaa