thonny/thonny #1920: Import crashes debugging.

harrkev, 2021-07-30:
I am trying to learn how the cryptography package works, and I encountered this....
Version 3.3.11

Enable "allow stepping into libraries" from the "Run & Debug" tab of options.
Enter this one-line program and save it:

from cryptography.hazmat.primitives import hashes

Click on "debug" and step over that one line (F6 button).  Get this:

Traceback (most recent call last):
  File "C:\Users\C40474\Desktop\example.py", line 1, in <module>
    from cryptography.hazmat.primitives import hashes
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 724, in exec_module
  File "<frozen importlib._bootstrap_external>", line 860, in get_code
IndexError: list index out of range

If you just run this, or disable stepping into libraries, then you don't get this error.  I just wanted to see how this package works on the inside...

aivarannamaa, 2021-07-31:
Thank you for the report!

Could you please check, whether the same problem occurs with "faster" debugger (Shift-F5)?

aivarannamaa, 2021-07-31:
It looks like someone has had similar problem: https://stackoverflow.com/questions/68522362/error-importing-custom-python-package-while-using-thonny-debug

harrkev, 2021-08-02:
Running "Shift-F5" seems to work just fine.  I can step into the code.  The
error only occurs running "Ctrl-F5".