thonny/thonny #1921: CircuitPython 7 Beta. Device not connecting?

bitboy85, 2021-07-31:
I'm currently playing around with CP 7 Beta and noticed, that thonny is not able to connect to the device. It outputs only a blank line on "Stop/restart". To get it working you must press CTRL + C

[image]

aivarannamaa, 2021-07-31:
Thank you for the report!

Which Thonny version are you using? On which platform? Did CP 6.3.0 work properly in the same Thonny installation?

bitboy85, 2021-08-02:
Hi, i started again with another board and wasn't able to reproduce the issue.  Maybe the issue is caused by the connected hardware or used libraries on the other board.
So i close this issue.

closed 2021-08-02 by bitboy85