thonny/thonny #1923: thonny.plugins.micropython.backend: Crash in backend

rfafuentes, 2021-08-01:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 135, in __init__
    self._prepare_after_soft_reboot(clean)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 157, in _prepare_after_soft_reboot
    self._update_cwd()
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/bare_metal_backend.py", line 390, in _update_cwd
    super()._update_cwd()
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 480, in _update_cwd
    self._cwd = self._evaluate("__thonny_helper.getcwd()")
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 567, in _evaluate
    raise ManagementError(script, out, err)
thonny.plugins.micropython.backend.ManagementError: Problem with a management command

SCRIPT:
__thonny_helper.print_mgmt_value(__thonny_helper.getcwd())

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 44, in getcwd
AttributeError: no such attribute

aivarannamaa, 2021-08-01:
Thank you for the report!

With which device did you get this error? Which Thonny version?

jnko, 2021-08-31:
Thonny 3.3.14, set to MicroPython (ESP32), Port: Auto
Python 3.9.6 (local), MicroPython 1.16 (device)
Device: DFRobot FireBeetle ESP32 V4.0 (DFR0478)

Very similar problem here. Unable to connect or run current script on the device.
`Device is busy -- can't perform this action now. Please wait or cancel work and try again!`

Pressing the boards reset button leads to:
```ERROR   thonny.plugins.micropython.backend: Crash in backend
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/thonny/plugins/micropython/backend.py", line 136, in __init__
    self._prepare_after_soft_reboot(clean)
  File "/usr/lib/python3.9/site-packages/thonny/plugins/micropython/backend.py", line 169, in _prepare_after_soft_reboot
    self._sys_path = self._fetch_sys_path()
  File "/usr/lib/python3.9/site-packages/thonny/plugins/micropython/backend.py", line 442, in _fetch_sys_path
    return self._evaluate("__thonny_helper.sys.path")
  File "/usr/lib/python3.9/site-packages/thonny/plugins/micropython/backend.py", line 611, in _evaluate
    raise ManagementError(script, out, err)
thonny.plugins.micropython.backend.ManagementError: Problem with a management command

SCRIPT:
__thonny_helper.print_mgmt_value(__thonny_helper.sys.path)

STDOUT:
<thonny>['', '/lib']</thonny>

STDERR:
>ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5640
load:0x40078000,len:12696
load:0x40080400,len:4292
entry 0x400806b0
MicroPython v1.16 on 2021-06-23; ESP32 module with ESP32
Type "help()" for more information.

Backend terminated or disconnected. Use 'Stop/Restart' to restart.```

Notable to say that other boards are working without problems, for example HUZZAH32.
Anyway I'm able to connect to the FireBeetle board within a terminal (ttyUSB0) without any problems.
Using other (oder) MicryPython versions does not help.

aivarannamaa, 2021-09-16:
Related to #1964

danscheidt, 2021-09-17:
I am having the exact same problem.

Using Thonny 3.3.10 on an rpi4 and Micropython 1.17 on an esp32-wrover. I can connect fine in a terminal using minicom but not Thonny.

jomasnash, 2021-10-31:
I also am having the exact same problem.
ESP-07 (512kb) with Micropython 1.17 (latest). to MicroPython (ESP32), Port: Auto
Thonny: 3.3.13  set to MicroPython (ES8266)

When using Putty, everything works. ESP-07 starts and show prompt >>>
the 'work-around:  (put 'submit_mode = paste' in [ESP8266] does not help.

jomasnash, 2021-11-04:
@aivarannamaa

I found the cause of this crash.

In my case I was using a micropython firmware that did not have the command 'getcwd' in module os.
To check:

Connect your module to Putty or minicom or another terminal program.
Check if micropython starts up (give prompt >>>).
If so, run the following commands:

`import os`
`dir(os)`

Now you should see 'getcwd' in the list that appears.
If not, than this is the cause of the 'crash'

aivarannamaa, 2021-11-05:
> I was using a micropython firmware that did not have the command 'getcwd' in module os

This is a news for me that a bare-metal port can omit `getcwd` (I knew that it's not present in Unix port).

@jomasnash, can you please make another experiment to be sore -- please check for this attribute via Putty just after hard-resetting your device. In some cases (after meeting errors at the class level) MicroPython can get to the state where namespaces are broken.

jomasnash, 2021-11-05:
> > I was using a micropython firmware that did not have the command 'getcwd' in module os
> 
> This is a news for me that a bare-metal port can omit `getcwd` (I knew that it's not present in Unix port).
> 
> @jomasnash, can you please make another experiment to be sure -- please check for this attribute via Putty just after hard-resetting your device. In some cases (after meeting errors at the class level) MicroPython can get to the state where namespaces are broken.

I did the test by hard resetting. After some search, it appears that firmware with 512k flash don't have a filesystem on it at all. (at least for the ESP8266). So all file related functions 'open' 'listdir' 'getcwd' etc. are missing in module os. For other ports this can also happen if you compile micropython yourself and do not set (or forget to set) the flags to include the filesystem.

May be a test on the existence of 'getcwd' can give a more clear error message from thonny.

aivarannamaa, 2021-11-05:
Thanks!

I'll see what can be done.