thonny/thonny #1930: Connect to WebREPL without restarting backend

JennaSys, 2021-08-02:
This is MicroPython/ESP8266 related, but as a potential feature addition, would it be possible to add an option to the Run menu to connect via WebREPL without restarting the backend?

My current workaround for this is to go into `Run` -> `Select - interpreter...` then dirty the config by re-selecting WebREPL, and clicking `OK`.  Thonny then connects to the device without resetting it, in the same way that the MicroPython WebREPL works by default.

Or let me know if there's another way to do it that I'm missing.

aivarannamaa, 2021-08-03:
Do I understand correctly, that you want to reconnect to WebREPL after losing the connection? In this case I suggest to try the back-end switcher in the lower-right corner of Thonny's main window -- just open it and click again on the desired item, even if it already has check-mark

JennaSys, 2021-08-03:
Correct, that is what I meant, and I had no idea that back-end switcher was even there.  I tried it and it does exactly what I want.
Thanks!

closed 2021-08-03 by JennaSys