thonny/thonny #1932: cannot same script into pico

artemskorsky, 2021-08-04:
I can run the script , but the. I place save button I am getting this error:

Could not write next block after having written 0 bytes to /main.py
ERROR   thonny.backend: PROBLEM WITH THONNY'S BACK-END
Error Problem with a management command

any solution for that?

artemskorsky, 2021-08-05:
@aivarannamaa do you have any idea how can I solve this issue?

aivarannamaa, 2021-08-05:
Are you using MicroPython or CircuitPython?

Do you have any files on the device? This can be checked via Thonny's file browser (View => Files).

Also, please check what does Thonny say about the storage space o your device (the "hamburger" menu next to "Raspberry Pi Pico" in the file browser):

[image]

Blahri, 2021-08-12:
I have the same issue and it is super annoying. For 8 of 10 attempts, when I try to save the code on the Pico (running Thonny on Windows 10), it gets stuck in the process and there is nothing I can do other than restart the Pico.  I've checked the storage on device, and I still have around 1.1 MB of free space. The script is around 17 KB, but it happened with smaller as well.

artemskorsky, 2021-08-12:
after investigating I found out that problem was on the memory chip on pico. I changed the pico, everything works fine

Blahri, 2021-08-15:
@aivarannamaa it wasn't exactly the same error code but whenever I tried to upload the code it got stuck, the Pico freezed and I had to power cycle it. Eventually I found out (suspect) that the size of the code was maybe just too much for the Pico? When I removed some unnecessary part of the code, it started to work fine.

aivarannamaa, 2021-08-21:
@Blahri, how big your code was when you were having problems? How many lines and/or bytes?

Blahri, 2021-08-21:
@aivarannamaa The module I had trouble uploading had 530 lines at that time and circa 17 kB.

aivarannamaa, 2023-07-05:
Have you had better experience with newer Thonny versions (e.g. 4.1.1)?