thonny/thonny #1934: COPY and PASTE bug

neil9, 2021-08-07:
Frequently - If i copy and paste or cut and paste (using ctrl-c  and ctrl_v) some lines to another location in code, thonny will randomly insert that code in other unwanted code space and in many places. Usually this causes a syntax or indentation error and can be fixed, but sometimes it will kill the code, and can take some real work in fixing it
It is random and is such that I avoid doing cut/pastes unless really necessary because of this issue.

aivarannamaa, 2021-08-08:
Thank you for the report!

Do I understand correctly that when pasting code, the code will be pasted somewhere else in addition to the place of the editing cursor? In these cases, does the text always get pasted at cursor as well?

Does it happen in the editor or shell or both?

Which platform and Thonny version are you using?

aivarannamaa, 2021-11-06:
Is your issue similar to #2036?

aivarannamaa, 2022-03-23:
Please see whether https://github.com/thonny/thonny/releases/tag/v4.0.0b1 works for you

aivarannamaa, 2023-01-29:
Please reopen if you still have this problem in Thonny 4.0.2

closed 2023-01-29 by aivarannamaa