thonny/thonny #1937: Flashing light and Thonny can't find the device.

NetDoc2k, 2021-08-11:
I have no idea how to troubleshoot this problem. Is there any direction or website I can check out.

aivarannamaa, 2021-08-11:
You have given too little details. Which device? How did you try to make Thonny see it? Does your device have MicroPython or CircuitPython installed? Which platform and Thonny version are you using?

NetDoc2k, 2021-08-11:
Thanks for the interest... I installed micropython yesterday: 3.9, I believe. It's the latest Thonny as of yesterday as well(3.3.14). I have nothing installed on the device yet that I know of. I have configured the interpreter to MicroPython (ESP32) and I get this error message...

Couldn't find the device automatically. 
Check the connection (making sure the device is not in bootloader mode) or choose
"Configure interpreter" in the interpreter menu (bottom-right corner of the window)
to select specific port or another interpreter.

NetDoc2k, 2021-08-11:
To take this further... how I know what port my device is on? 

The device is an ESP-WROOM-32 found here: https://smile.amazon.com/gp/product/B077KJNVFP/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

NetDoc2k, 2021-08-11:
OK, recovering my steps, I have stopped the flashing. I was stuck on changing/configuring the interpreter. There were two main choices. One said ESP32 and the other said Python 3.7.9 (bundled). I tried that and the blue light is no on solid. Progress? I'll try to continue with the tutorial I'm on and see.

aivarannamaa, 2021-08-11:
If you haven't installed MicroPython yet, then download firmware from https://micropython.org/download/esp32/

Then click on Thonny's back-end switcher button (lower-right corner of the main window), select "Configure interpreter" => "MicroPython (ESP-32)" => "Install or update firmware" => select port => select firmware => start flashing

If you don't know the port, then you can use the "Reload" button in the firmware installation dialog -- unplug your device and see which port disappears.

NetDoc2k, 2021-08-11:
OK, that's not it, either. I apologize for the rambling here... I'm trying `different options and recording their results here. 

I tried a different USB cord, and that caused my PC to install it as a device.`On checking my Device manager, I see it hasn't been able to install the correct driver. It's a CP2104 USB to UART Controller.

NetDoc2k, 2021-08-11:
When I install the driver, the device shows up, then drops out, then shows up, then drops out and beeps all the time.

closed 2021-11-06 by aivarannamaa