thonny/thonny #1939: Can't access MicroPython from Thonny

sk8nfool, 2021-08-13:
I installed Thonny using several of the methods described on your webpage. Nothing seems to work quite right. In all cases, when I click on the version number in the lower right corner of the thonny window I don't get MicroPython anything.  Is there some plugin I'm missing or is the menu under the version number dynamic (I don't have my RPi-Pico plugged in)?
I'm running Thonny 3.6.9 on Ubuntu 18.04 on my 64 bit laptop.

aivarannamaa, 2021-08-13:
Thank you for the report!

> when I click on the version number in the lower right corner of the thonny window I don't get MicroPython anything.

MicroPython will be only available if you have a MicroPython device plugged in to a USB port. If you want to check the possible back-ends, then select "Run => Select interpreter".

> I'm running Thonny 3.6.9

Please re-check this. Latest Thonny version is 3.3.14

sk8nfool, 2021-08-14:
My mistake! Thonny's about box says exactly that. It's Python that is
version 3.6.9. My apologies.

aivarannamaa, 2021-11-06:
Sorry, I've somehow missed your reply. Have you found solution to your problem?

closed 2022-03-23 by aivarannamaa