thonny/thonny #1942: Auto Closing Brackets and Quotes

Sushi2k6, 2021-08-20:
I have been using thonny for the past 3 months, and I'd give it a 4/5 (according to me). I want my IDE (or editor) to be simple, accessible and quick, basically it should be **lightweigh**t. An that's what thonny is. But with a few setbacks, like

-> Auto closing Brackets and Quotes
-> Ability to install packages from different sources
->Better Assistant

If these are implemented in thonny, It'd be a lot better for everyone
Hope it will. (cuz I cant find anything that suites me, and the closest one is thonny!)

aivarannamaa, 2021-08-21:
Thank you for the suggestions!

Auto-closing parens in in my TODO list: #59

> Ability to install packages from different sources

Do you mean specifying different or extra index instead or besides PyPI?

> Better Assistant

What do you mean specifically?

Sushi2k6, 2021-09-03:
Besides PyPI.
And, Assistant can give definitions, indicate errors on code while typing, this makes it so much easier.

BTW, The closing parens doesn't include [] and {} right? It'd be better if the quote was also closed automatically

ser9ee, 2021-09-25:
> Thank you for the suggestions!
> 
> Auto-closing parens in in my TODO list: #59
> 
> > Ability to install packages from different sources
> 
> Do you mean specifying different or extra index instead or besides PyPI?
> 
> > Better Assistant
> 
> What do you mean specifically?

Hello.
Thonny is very good and i like it so much, but auto-closing is soooo important for usability. please add this option as quickly as possible. Thank you for your great application!
(), [], {}, '', ""

Sushi2k6, 2022-03-23:
Hello! Is the issue solved yet? It hasn't been added yet.

aivarannamaa, 2022-03-23:
No, it's still in the TODO list and unfortunately there are lots of more important issue before this.

Sushi2k6, 2022-03-24:
oh ok. Is there any possible way to see the TODO list?

aivarannamaa, 2022-03-24:
Basically https://github.com/thonny/thonny/issues is the TODO list. You can filter by the milestone to get a more structured picture.

Sushi2k6, 2022-03-24:
Oh ok, thanks. Actually I was confused by this 
[image]

RushilJalal, 2022-08-02:
When is the autoclosing quotes and brackets going to be implemented? Thanks

aivarannamaa, 2022-08-02:
> When is the autoclosing quotes and brackets going to be implemented?

I can't promise any date. When more import issues are fixed and I have time for this.

visualearner, 2026-09-22:
Hello, does Thonny have Auto Closing Brackets and Quotes yet? I haven't been able to find an answer but it would be so much smoother if it had it! thank you

aivarannamaa, 2026-10-26:
No, not yet

chagraoui1903, 2026-11-02:
No Way This is still not fixed