thonny/thonny #1943: script sended with RUN command is not complete

favnec5, 2021-08-20:
hi,
I try to run script with 721 lines on my M5Core2 micropython 1.15 board.
But the "Run" function split source or something like that :

[image]

I tested Thonny version 3.3.11, 3.3.13 for window (not the 3.3.14).
An other ide with the same script work.
Do you know this problem ?
best reguards,
Thomas

aivarannamaa, 2021-08-21:
Thank you for the report!

It looks like a byte gets lost when Thonny sends code to the device. When you try to run the same script several times, does the problem happen always at the same spot?

Are you using the official M5 firmware?

favnec5, 2021-08-21:
Nop this is pure micropython 1.14 and yes the same byte every Time. If i change the file size, the byte change. Sometimes i can put comment at the right place to work.

aivarannamaa, 2021-08-22:
Could you send me the whole script? I don't have a Core2, but maybe I can reproduce it with the original M5Stack.

If you are willing to do some tweaking, then please read https://github.com/thonny/thonny/wiki/MicroPython#advanced-configuration and try different values for `submit_mode`, `write_block_size` and `write_block_delay`. The default values are `raw_paste`, `255` and `0.01`.

favnec5, 2021-08-24:
Hi aivarannamaa 
Thanks for your help. **Submit_mode=raw** seems to solve my problem ! 
it's crazy that I didn't see that.
I can't send you my script because it use lot of things (ble beacons, pictures on the board..., wav files...), it coulnd't work.
it's a good news for me, i love working with Thonny :-)

aivarannamaa, 2022-05-18:
I could reproduce this but did not find any way to overcome this problem other than using raw REPL submission mode. Hopefully M5 starts using newer MicroPython versions soon (https://community.m5stack.com/topic/4096/uiflow-2-0-2-firmware-preview-version), which support faster raw-paste mode, which is used as default in Thonny.

closed 2022-05-18 by aivarannamaa